tukl-msd/DRAMSys #1: Error in build the DRAMSys

explorer1022, 2020-07-20:
hello
when I try to build the DRAMSys it shows me an errors message 

ahmad@ahmad-VirtualBox:~/DRAMSys/DRAMSys$ cmake /home/ahmad/DRAMSys/DRAMSys
-- The C compiler identification is GNU 9.3.0
-- The CXX compiler identification is GNU 9.3.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
CMake Error at library/CMakeLists.txt:55 (add_subdirectory):
  The source directory

    /home/ahmad/DRAMSys/DRAMSys/library/src/common/third_party/DRAMPower

  does not contain a CMakeLists.txt file.

CMake Error at library/CMakeLists.txt:58 (add_subdirectory):
  The source directory

    /home/ahmad/DRAMSys/DRAMSys/library/src/common/third_party/nlohmann

  does not contain a CMakeLists.txt file.

CMake Error at library/CMakeLists.txt:69 (add_subdirectory):
  The source directory

    /home/ahmad/DRAMSys/DRAMSys/library/src/common/third_party/systemc

  does not contain a CMakeLists.txt file.

---- Building with SystemC submodule
-- Configuring incomplete, errors occurred!
See also "/home/farhan/DRAMSys/DRAMSys/CMakeFiles/CMakeOutput.log".

myzinsky, 2020-07-20:
Have you cloned dramsys with the ```--recursive``` option?

explorer1022, 2020-07-20:
How I can clone it with  --recursive option.

myzinsky, 2020-07-20:
```git clone --recursive https://github.com/tukl-msd/DRAMSys.git```

@lsteiner-tukl can you add this to the readme?

explorer1022, 2020-07-20:
now it works with me.
many thanks.

closed 2020-07-20 by lsteiner-tukl