tukl-msd/DRAMSys #2: error in makefile

explorer1022, 2020-07-21:
hello 
when I do  make SYSTEMC_WRAPPER=y
I got the bellow error, how I can solve it?  

cd ./flex    ; make ;
make[1]: Entering directory '/home/ ahmad/3d-ice-latest/3d-ice-2.2.6/flex'
make[1]: Nothing to be done for 'all'.
make[1]: Leaving directory '/home/ ahmad/3d-ice-latest/3d-ice-2.2.6/flex'
cd ./bison   ; make ;
make[1]: Entering directory '/home/ ahmad/3d-ice-latest/3d-ice-2.2.6/bison'
make[1]: Nothing to be done for 'all'.
make[1]: Leaving directory '/home/ahmad/3d-ice-latest/3d-ice-2.2.6/bison'
cd ./sources ; make ;
make[1]: Entering directory '/home/ahmad/3d-ice-latest/3d-ice-2.2.6/sources'
gcc -MD -O3 -Wall -Wextra -Werror -I../include -I/home/ahmad/SuperLU_4.3/SRC -c ../sources/dimensions.c -o ../sources/dimensions.o
../sources/dimensions.c: In function ‘compute_number_of_connections’:
../sources/dimensions.c:345:17: error: this statement may fall through [-Werror=implicit-fallthrough=]
  345 |             tmp += 2 ;
      |             ~~~~^~~~
../sources/dimensions.c:347:9: note: here
  347 |         case TDICE_CHANNEL_MODEL_PF_INLINE :
      |         ^~~~
cc1: all warnings being treated as errors
make[1]: *** [../makefile.def:181: ../sources/dimensions.o] Error 1
make[1]: Leaving directory '/home/ahmad/3d-ice-latest/3d-ice-2.2.6/sources'
make: *** [Makefile:50: lib] Error 2

lsteiner-tukl, 2020-07-21:
Try to comment out line 45 (CFLAGS := $(CFLAGS) -Wall -Wextra -Werror) in the sources/Makefile.

explorer1022, 2020-07-21:
After commenting out line 45 it shows me like this:

ahmad@ahmad-VirtualBox:~/3d-ice-latest/3d-ice-2.2.6$ make SYSTEMC_WRAPPER=y
cd ./flex    ; make ;
make[1]: Entering directory '/home/ahmad/3d-ice-latest/3d-ice-2.2.6/flex'
make[1]: Nothing to be done for 'all'.
make[1]: Leaving directory '/home/ahmad/3d-ice-latest/3d-ice-2.2.6/flex'
cd ./bison   ; make ;
make[1]: Entering directory '/home/ahmad/3d-ice-latest/3d-ice-2.2.6/bison'
make[1]: Nothing to be done for 'all'.
make[1]: Leaving directory '/home/ahmad/3d-ice-latest/3d-ice-2.2.6/bison'
cd ./sources ; make ;
make[1]: Entering directory '/home/ahmad/3d-ice-latest/3d-ice-2.2.6/sources'
gcc -MD -O3 -Wall -Wextra -Werror -I../include -I/home/ahmad/SuperLU_4.3/SRC -c ../sources/dimensions.c -o ../sources/dimensions.o
../sources/dimensions.c: In function ‘compute_number_of_connections’:
../sources/dimensions.c:345:17: error: this statement may fall through [-Werror=implicit-fallthrough=]
  345 |             tmp += 2 ;
      |             ~~~~^~~~
../sources/dimensions.c:347:9: note: here
  347 |         case TDICE_CHANNEL_MODEL_PF_INLINE :
      |         ^~~~
cc1: all warnings being treated as errors
make[1]: *** [../makefile.def:181: ../sources/dimensions.o] Error 1
make[1]: Leaving directory '/home/ahmad/3d-ice-latest/3d-ice-2.2.6/sources'
make: *** [Makefile:50: lib] Error 2
ahmad@ahmad-VirtualBox:~/3d-ice-latest/3d-ice-2.2.6$ make SYSTEMC_WRAPPER=y
cd ./flex    ; make ;
make[1]: Entering directory '/home/ahmad/3d-ice-latest/3d-ice-2.2.6/flex'
make[1]: Nothing to be done for 'all'.
make[1]: Leaving directory '/home/ahmad/3d-ice-latest/3d-ice-2.2.6/flex'
cd ./bison   ; make ;
make[1]: Entering directory '/home/ahmad/3d-ice-latest/3d-ice-2.2.6/bison'
make[1]: Nothing to be done for 'all'.
make[1]: Leaving directory '/home/ahmad/3d-ice-latest/3d-ice-2.2.6/bison'
cd ./sources ; make ;
make[1]: Entering directory '/home/ahmad/3d-ice-latest/3d-ice-2.2.6/sources'
gcc -MD -O3 -I../include -I/home/ahmad/SuperLU_4.3/SRC -c ../sources/dimensions.c -o ../sources/dimensions.o
gcc -MD -O3 -I../include -I/home/ahmad/SuperLU_4.3/SRC -c ../sources/floorplan_element.c -o ../sources/floorplan_element.o
gcc -MD -O3 -I../include -I/home/ahmad/SuperLU_4.3/SRC -c ../sources/floorplan_element_list.c -o ../sources/floorplan_element_list.o
gcc -MD -O3 -I../include -I/home/ahmad/SuperLU_4.3/SRC -c ../sources/floorplan_file_parser.c -o ../sources/floorplan_file_parser.o
gcc -MD -O3 -I../include -I/home/ahmad/SuperLU_4.3/SRC -c ../sources/floorplan_matrix.c -o ../sources/floorplan_matrix.o
gcc -MD -O3 -I../include -I/home/ahmad/SuperLU_4.3/SRC -c ../sources/floorplan.c -o ../sources/floorplan.o
gcc -MD -O3 -I../include -I/home/ahmad/SuperLU_4.3/SRC -c ../sources/heat_sink.c -o ../sources/heat_sink.o
gcc -MD -O3 -I../include -I/home/ahmad/SuperLU_4.3/SRC -c ../sources/ic_element.c -o ../sources/ic_element.o
gcc -MD -O3 -I../include -I/home/ahmad/SuperLU_4.3/SRC -c ../sources/ic_element_list.c -o ../sources/ic_element_list.o
gcc -MD -O3 -I../include -I/home/ahmad/SuperLU_4.3/SRC -c ../sources/inspection_point.c -o ../sources/inspection_point.o
gcc -MD -O3 -I../include -I/home/ahmad/SuperLU_4.3/SRC -c ../sources/inspection_point_list.c -o ../sources/inspection_point_list.o
gcc -MD -O3 -I../include -I/home/ahmad/SuperLU_4.3/SRC -c ../sources/layer.c -o ../sources/layer.o
gcc -MD -O3 -I../include -I/home/ahmad/SuperLU_4.3/SRC -c ../sources/layer_list.c -o ../sources/layer_list.o
gcc -MD -O3 -I../include -I/home/ahmad/SuperLU_4.3/SRC -c ../sources/layout_file_parser.c -o ../sources/layout_file_parser.o
gcc -MD -O3 -I../include -I/home/ahmad/SuperLU_4.3/SRC -c ../sources/material.c -o ../sources/material.o
gcc -MD -O3 -I../include -I/home/ahmad/SuperLU_4.3/SRC -c ../sources/material_list.c -o ../sources/material_list.o
gcc -MD -O3 -I../include -I/home/ahmad/SuperLU_4.3/SRC -c ../sources/material_element.c -o ../sources/material_element.o
gcc -MD -O3 -I../include -I/home/ahmad/SuperLU_4.3/SRC -c ../sources/material_element_list.c -o ../sources/material_element_list.o
gcc -MD -O3 -I../include -I/home/ahmad/SuperLU_4.3/SRC -c ../sources/network_message.c -o ../sources/network_message.o
gcc -MD -O3 -I../include -I/home/ahmad/SuperLU_4.3/SRC -c ../sources/network_socket.c -o ../sources/network_socket.o
gcc -MD -O3 -I../include -I/home/ahmad/SuperLU_4.3/SRC -c ../sources/output.c -o ../sources/output.o
gcc -MD -O3 -I../include -I/home/ahmad/SuperLU_4.3/SRC -c ../sources/power_grid.c -o ../sources/power_grid.o
gcc -MD -O3 -I../include -I/home/ahmad/SuperLU_4.3/SRC -c ../sources/powers_queue.c -o ../sources/powers_queue.o
gcc -MD -O3 -I../include -I/home/ahmad/SuperLU_4.3/SRC -c ../sources/stack_description.c -o ../sources/stack_description.o
gcc -MD -O3 -I../include -I/home/ahmad/SuperLU_4.3/SRC -c ../sources/stack_element.c -o ../sources/stack_element.o
gcc -MD -O3 -I../include -I/home/ahmad/SuperLU_4.3/SRC -c ../sources/stack_element_list.c -o ../sources/stack_element_list.o
gcc -MD -O3 -I../include -I/home/ahmad/SuperLU_4.3/SRC -c ../sources/stack_file_parser.c -o ../sources/stack_file_parser.o
gcc -MD -O3 -I../include -I/home/ahmad/SuperLU_4.3/SRC -c ../sources/system_matrix.c -o ../sources/system_matrix.o
gcc -MD -O3 -I../include -I/home/ahmad/SuperLU_4.3/SRC -c ../sources/string_t.c -o ../sources/string_t.o
gcc -MD -O3 -I../include -I/home/ahmad/SuperLU_4.3/SRC -c ../sources/thermal_data.c -o ../sources/thermal_data.o
[... cut ...]
g++ -std=c++11 -O2 -Wall -Wextra -Werror -I../include -I/home/ahmad/SuperLU_4.3/SRC -I/home/ahmad/systemc-2.3.x/include -c ../sources/IceWrapper.cpp -o ../sources/IceWrapper.o
In file included from ../sources/IceWrapper.cpp:37:
../include/IceWrapper.h:44:10: fatal error: systemc.h: No such file or directory
   44 | #include <systemc.h>
      |          ^~~~~~~~~~~
compilation terminated.
make[1]: *** [../makefile.def:191: ../sources/IceWrapper.o] Error 1
make[1]: Leaving directory '/home/ahmad/3d-ice-latest/3d-ice-2.2.6/sources'
make: *** [Makefile:50: lib] Error 2

lsteiner-tukl, 2020-07-21:
It seems that you haven't properly set the variables in the makefile.def. Of course the variables in the readme are only an example.

SLU_MAIN = $(HOME)/SuperLU_$(SLU_VERSION)
YACC = bison-2.4.1
SYSTEMC_ARCH = linux64
SYSTEMC_MAIN = $(HOME)/systemc-2.3.x

I assume you are using the SystemC submodule, so the directory of your SystemC include files and library are not $(HOME)/systemc-2.3.x. 

If the DRAMSys repo is located in your home directory, you should set them as follows:

SYSTEMC_VERSION = 2.3.4
#SYSTEMC_ARCH    = linux
SYSTEMC_ARCH    = linux64
SYSTEMC_MAIN    = $(HOME)/DRAMSys/DRAMSys/library/src/common/third_party/systemc/src
SYSTEMC_INCLUDE = $(SYSTEMC_MAIN)
SYSTEMC_LIB     = $(HOME)/DRAMSys/build/library/src/common/third_party/systemc/src
SYSTEMC_LIBS    = -L$(SYSTEMC_LIB) -lsystemc

Before you try it out please remove 3D-ICE and download it again.

closed 2020-07-27 by myzinsky