tukl-msd/DRAMSys #3: make fails when building DRAMPower

sherylll, 2020-07-24:
System info: REHL, cmake version 3.18.0

after running cmake and make in the build directory, I got the following

```
Scanning dependencies of target DRAMPower
[ 35%] Building CXX object library/src/common/third_party/DRAMPower/CMakeFiles/DRAMPower.dir/src/libdrampower/LibDRAMPower.cc.o
[ 35%] Building CXX object library/src/common/third_party/DRAMPower/CMakeFiles/DRAMPower.dir/src/CommandAnalysis.cc.o
workspace/DRAMSys/DRAMSys/library/src/common/third_party/DRAMPower/src/CommandAnalysis.cc: In member function ‘void DRAMPower::CommandAnalysis::evaluateCommands(std::vector<DRAMPower::MemCommand>&)’:
workspace/DRAMSys/DRAMSys/library/src/common/third_party/DRAMPower/src/CommandAnalysis.cc:216:17: error: expected initializer before ‘:’ token
workspace/DRAMSys/DRAMSys/library/src/common/third_party/DRAMPower/src/CommandAnalysis.cc:292:1: error: expected primary-expression at end of input
workspace/DRAMSys/DRAMSys/library/src/common/third_party/DRAMPower/src/CommandAnalysis.cc:292:1: error: expected ‘;’ at end of input
workspace/DRAMSys/DRAMSys/library/src/common/third_party/DRAMPower/src/CommandAnalysis.cc:292:1: error: expected primary-expression at end of input
workspace/DRAMSys/DRAMSys/library/src/common/third_party/DRAMPower/src/CommandAnalysis.cc:292:1: error: expected ‘)’ at end of input
workspace/DRAMSys/DRAMSys/library/src/common/third_party/DRAMPower/src/CommandAnalysis.cc:292:1: error: expected statement at end of input
workspace/DRAMSys/DRAMSys/library/src/common/third_party/DRAMPower/src/CommandAnalysis.cc:292:1: error: expected ‘}’ at end of input
make[2]: *** [library/src/common/third_party/DRAMPower/CMakeFiles/DRAMPower.dir/src/CommandAnalysis.cc.o] Error 1
make[1]: *** [library/src/common/third_party/DRAMPower/CMakeFiles/DRAMPower.dir/all] Error 2
make: *** [all] Error 2
```
Any ideas?

lsteiner-tukl, 2020-07-24:
Can you make sure that DRAMPower is compiled with C++11 support? It seems that the auto keyword or the range-based for loop is causing the error.

sherylll, 2020-07-24:
This is part of the cmake output:
```
-- ------------------------------------------------------------------------
-- CMAKE_CXX_STANDARD = 11
-- CMAKE_CXX_STANDARD_REQUIRED = ON
-- CMAKE_SYSTEM = Linux-2.6.32-754.11.1.el6.x86_64
-- CMAKE_SYSTEM_PROCESSOR = x86_64
-- QT_ARCH = x86_64
-- Threads_FOUND = TRUE
-- CMAKE_USE_PTHREADS_INIT = 1
-- CMAKE_THREAD_LIBS_INIT = -pthread
-- ----------------------------------------------------------------------
```
looks c++11 is correctly selected.

My gcc version is 7.3.0.

I also tried to change all occurrence of std0x in the DRAMPower directory to std11, didn't help though...

lsteiner-tukl, 2020-07-24:
I'm not a cmake expert. What you can try out is to add the line

set(CMAKE_CXX_STANDARD_REQUIRED ON)

after 

set(CMAKE_CXX_STANDARD 11 CACHE STRING "C++ Version")

in the top level CMakeLists.txt of DRAMSys. This might solve the problem. Maybe you can also try to build DRAMPower separately and see if it works.

myzinsky, 2020-07-24:
I also would recommend to build DRAMPower as standalone first. I reccomend the development branch https://github.com/tukl-msd/DRAMPower/tree/develop. We should sort the problem there since its not a DRAMSys problem in the first place.

@sherylll can you give that a try?

sherylll, 2020-07-24:
Hi,

I checkout the develop branch of DRAMPower and built it first. Works fine.

However when run make under build for DRAMSys:
```
Scanning dependencies of target DRAMPower
[ 36%] Building CXX object library/src/common/third_party/DRAMPower/CMakeFiles/DRAMPower.dir/src/libdrampower/LibDRAMPower.cc.o
In file included from workspace/DRAMSys/DRAMSys/library/src/common/third_party/DRAMPower/src/jsonparser/JSONParser.h:46:0,
                 from 
workspace/DRAMSys/DRAMSys/library/src/common/third_party/DRAMPower/src/MemSpecParser.h:40,
                 from 
...........
workspace/DRAMSys/DRAMSys/library/src/common/third_party/DRAMPower/src/libdrampower/LibDRAMPower.cc:42:
workspace/DRAMSys/DRAMSys/library/src/common/third_party/DRAMPower/src/common/libraries/json/json.h:1721:14: error: #error "unsupported GCC version - see https://github.com/nlohmann/json#supported-compilers"
```
It seems to find the the wrong gcc compiler somehow? Running `gcc --version` in the current console returns
```
gcc (GCC) 7.3.0
```

lsteiner-tukl, 2020-07-24:
Can you try to export the CC and CXX variables that tell cmake which compiler to choose?

sherylll, 2020-07-25:
Hi thanks, setting the variables makes a difference!

Also I checkout the wrong version of DRAMPower, it has to be made with exactly the commit given.

Closing the issue now.

closed 2020-07-25 by sherylll