tukl-msd/DRAMSys #4: Leaving directory issue

explorer1022, 2020-07-24:
I am trying to run SuperLU_4.3 on the ubuntu 20.04 but it shows me make[1]: Leaving directory.

Can you help with this problem?

ahmad@ahmad-VirtualBox:~$ cd SuperLU_4.3/
ahmad@ahmad-VirtualBox:~/SuperLU_4.3$ cp MAKE_INC/make.linux make.inc
ahmad@ahmad-VirtualBox:~/SuperLU_4.3$ make superlulib
( cd SRC; make )
make[1]: Entering directory '/home/ahmad/SuperLU_4.3/SRC'
gcc -O3 -DAdd_ -DUSE_VENDOR_BLAS -c sgssv.c
gcc -O3 -DAdd_ -DUSE_VENDOR_BLAS -c sgssvx.c
gcc -O3 -DAdd_ -DUSE_VENDOR_BLAS -c ssp_blas2.c
ssp_blas2.c: In function ‘sp_strsv’:
ssp_blas2.c:155:7: warning: implicit declaration of function ‘strsv_’; did you mean ‘strsep’? [-Wimplicit-function-declaration]
  155 |       strsv_("L", "N", "U", &nsupc, &Lval[luptr], &nsupr,
      |       ^~~~~~
      |       strsep
ssp_blas2.c:158:7: warning: implicit declaration of function ‘sgemv_’ [-Wimplicit-function-declaration]
  158 |       sgemv_("N", &nrow, &nsupc, &alpha, &Lval[luptr+nsupc],
      |       ^~~~~~
gcc -O3 -DAdd_ -DUSE_VENDOR_BLAS -c ssp_blas3.c
gcc -O3 -DAdd_ -DUSE_VENDOR_BLAS -c sgscon.c
gcc -O3 -DAdd_ -DUSE_VENDOR_BLAS -c slangs.c
gcc -O3 -DAdd_ -DUSE_VENDOR_BLAS -c sgsequ.c
gcc -O3 -DAdd_ -DUSE_VENDOR_BLAS -c slaqgs.c
gcc -O3 -DAdd_ -DUSE_VENDOR_BLAS -c spivotgrowth.c
gcc -O3 -DAdd_ -DUSE_VENDOR_BLAS -c sgsrfs.c
gcc -O3 -DAdd_ -DUSE_VENDOR_BLAS -c sgstrf.c
gcc -O3 -DAdd_ -DUSE_VENDOR_BLAS -c sgstrs.c
sgstrs.c: In function ‘sgstrs’:
sgstrs.c:191:3: warning: implicit declaration of function ‘strsm_’; did you mean ‘strsep’? [-Wimplicit-function-declaration]
  191 |   strsm_("L", "L", "N", "U", &nsupc, &nrhs, &alpha,
      |   ^~~~~~
      |   strsep
sgstrs.c:194:3: warning: implicit declaration of function ‘sgemm_’ [-Wimplicit-function-declaration]
  194 |   sgemm_( "N", "N", &nrow, &nrhs, &nsupc, &alpha,
      |   ^~~~~~
gcc -O3 -DAdd_ -DUSE_VENDOR_BLAS -c scopy_to_ucol.c
gcc -O3 -DAdd_ -DUSE_VENDOR_BLAS -c ssnode_dfs.c
gcc -O3 -DAdd_ -DUSE_VENDOR_BLAS -c ssnode_bmod.c
ssnode_bmod.c: In function ‘ssnode_bmod’:
ssnode_bmod.c:97:2: warning: implicit declaration of function ‘strsv_’; did you mean ‘strsep’? [-Wimplicit-function-declaration]
   97 |  strsv_( "L", "N", "U", &nsupc, &lusup[luptr], &nsupr,
      |  ^~~~~~
      |  strsep
ssnode_bmod.c:99:2: warning: implicit declaration of function ‘sgemv_’ [-Wimplicit-function-declaration]
   99 |  sgemv_( "N", &nrow, &nsupc, &alpha, &lusup[luptr+nsupc], &nsupr,
      |  ^~~~~~
gcc -O3 -DAdd_ -DUSE_VENDOR_BLAS -c spanel_dfs.c
gcc -O3 -DAdd_ -DUSE_VENDOR_BLAS -c spanel_bmod.c
spanel_bmod.c: In function ‘spanel_bmod’:
spanel_bmod.c:224:7: warning: implicit declaration of function ‘strsv_’; did you mean ‘strsep’? [-Wimplicit-function-declaration]
  224 |       strsv_( "L", "N", "U", &segsze, &lusup[luptr],
      |       ^~~~~~
      |       strsep
spanel_bmod.c:271:7: warning: implicit declaration of function ‘sgemv_’ [-Wimplicit-function-declaration]
  271 |       sgemv_("N", &block_nrow, &segsze, &alpha, &lusup[luptr1],
      |       ^~~~~~
gcc -O3 -DAdd_ -DUSE_VENDOR_BLAS -c sreadhb.c
sreadhb.c: In function ‘sReadValues’:
sreadhb.c:157:2: warning: ignoring return value of ‘fgets’, declared with attribute warn_unused_result [-Wunused-result]
  157 |  fgets(buf, 100, fp);    /* read a line at a time */
      |  ^~~~~~~~~~~~~~~~~~~
sreadhb.c: In function ‘sreadhb’:
sreadhb.c:187:5: warning: ignoring return value of ‘fgets’, declared with attribute warn_unused_result [-Wunused-result]
  187 |     fgets(buf, 100, fp);
      |     ^~~~~~~~~~~~~~~~~~~
sreadhb.c:199:2: warning: ignoring return value of ‘fscanf’, declared with attribute warn_unused_result [-Wunused-result]
  199 |  fscanf(fp, "%14c", buf); buf[14] = 0;
      |  ^~~~~~~~~~~~~~~~~~~~~~~
[... 749 lines cut ...]
zlacon.c:157:5: warning: implicit declaration of function ‘zcopy_’ [-Wimplicit-function-declaration]
  157 |     zcopy_(n, x, &c__1, v, &c__1);
      |     ^~~~~~
gcc -O3 -DAdd_ -DUSE_VENDOR_BLAS -c dzsum1.c
gcc -O3 -DAdd_ -DUSE_VENDOR_BLAS -c izmax1.c
ar cr /home/ahmad/SuperLU_4.3/lib/libsuperlu_4.3.a \
dcomplex.o zgssv.o zgssvx.o zsp_blas2.o zsp_blas3.o zgscon.o zlangs.o zgsequ.o zlaqgs.o zpivotgrowth.o zgsrfs.o zgstrf.o zgstrs.o zcopy_to_ucol.o zsnode_dfs.o zsnode_bmod.o zpanel_dfs.o zpanel_bmod.o zreadhb.o zreadrb.o zreadtriple.o zcolumn_dfs.o zcolumn_bmod.o zpivotL.o zpruneL.o zmemory.o zutil.o zmyblas2.o zgsisx.o zgsitrf.o zldperm.o ilu_zdrop_row.o ilu_zsnode_dfs.o ilu_zcolumn_dfs.o ilu_zpanel_dfs.o ilu_zcopy_to_ucol.o ilu_zpivotL.o zdiagonal.o superlu_timer.o util.o memory.o get_perm_c.o mmd.o sp_coletree.o sp_preorder.o sp_ienv.o relax_snode.o heap_relax_snode.o colamd.o ilu_relax_snode.o ilu_heap_relax_snode.o mark_relax.o mc64ad.o qselect.o lsame.o xerbla.o zlacon.o dzsum1.o izmax1.o dlamch.o
ranlib /home/ahmad/SuperLU_4.3/lib/libsuperlu_4.3.a
make[1]: Leaving directory '/home/ahmad/SuperLU_4.3/SRC'
ahmad@ahmad-VirtualBox:~/SuperLU_4.3$ make superlulib
( cd SRC; make )
make[1]: Entering directory '/home/ahmad/SuperLU_4.3/SRC'
ar cr /home/ahmad/SuperLU_4.3/lib/libsuperlu_4.3.a \
sgssv.o sgssvx.o ssp_blas2.o ssp_blas3.o sgscon.o slangs.o sgsequ.o slaqgs.o spivotgrowth.o sgsrfs.o sgstrf.o sgstrs.o scopy_to_ucol.o ssnode_dfs.o ssnode_bmod.o spanel_dfs.o spanel_bmod.o sreadhb.o sreadrb.o sreadtriple.o scolumn_dfs.o scolumn_bmod.o spivotL.o spruneL.o smemory.o sutil.o smyblas2.o sgsisx.o sgsitrf.o sldperm.o ilu_sdrop_row.o ilu_ssnode_dfs.o ilu_scolumn_dfs.o ilu_spanel_dfs.o ilu_scopy_to_ucol.o ilu_spivotL.o sdiagonal.o superlu_timer.o util.o memory.o get_perm_c.o mmd.o sp_coletree.o sp_preorder.o sp_ienv.o relax_snode.o heap_relax_snode.o colamd.o ilu_relax_snode.o ilu_heap_relax_snode.o mark_relax.o mc64ad.o qselect.o lsame.o xerbla.o slacon.o  slamch.o
ranlib /home/ahmad/SuperLU_4.3/lib/libsuperlu_4.3.a
ar cr /home/ahmad/SuperLU_4.3/lib/libsuperlu_4.3.a \
                dgssv.o dgssvx.o dsp_blas2.o dsp_blas3.o dgscon.o dlangs.o dgsequ.o dlaqgs.o dpivotgrowth.o dgsrfs.o dgstrf.o dgstrs.o dcopy_to_ucol.o dsnode_dfs.o dsnode_bmod.o dpanel_dfs.o dpanel_bmod.o dreadhb.o dreadrb.o dreadtriple.o dcolumn_dfs.o dcolumn_bmod.o dpivotL.o dpruneL.o dmemory.o dutil.o dmyblas2.o dgsisx.o dgsitrf.o dldperm.o ilu_ddrop_row.o ilu_dsnode_dfs.o ilu_dcolumn_dfs.o ilu_dpanel_dfs.o ilu_dcopy_to_ucol.o ilu_dpivotL.o ddiagonal.o  superlu_timer.o util.o memory.o get_perm_c.o mmd.o sp_coletree.o sp_preorder.o sp_ienv.o relax_snode.o heap_relax_snode.o colamd.o ilu_relax_snode.o ilu_heap_relax_snode.o mark_relax.o mc64ad.o qselect.o lsame.o xerbla.o dlacon.o dlamch.o
ranlib /home/ahmad/SuperLU_4.3/lib/libsuperlu_4.3.a
ar cr /home/ahmad/SuperLU_4.3/lib/libsuperlu_4.3.a \
scomplex.o cgssv.o cgssvx.o csp_blas2.o csp_blas3.o cgscon.o clangs.o cgsequ.o claqgs.o cpivotgrowth.o cgsrfs.o cgstrf.o cgstrs.o ccopy_to_ucol.o csnode_dfs.o csnode_bmod.o cpanel_dfs.o cpanel_bmod.o creadhb.o creadrb.o creadtriple.o ccolumn_dfs.o ccolumn_bmod.o cpivotL.o cpruneL.o cmemory.o cutil.o cmyblas2.o cgsisx.o cgsitrf.o cldperm.o ilu_cdrop_row.o ilu_csnode_dfs.o ilu_ccolumn_dfs.o ilu_cpanel_dfs.o ilu_ccopy_to_ucol.o ilu_cpivotL.o cdiagonal.o superlu_timer.o util.o memory.o get_perm_c.o mmd.o sp_coletree.o sp_preorder.o sp_ienv.o relax_snode.o heap_relax_snode.o colamd.o ilu_relax_snode.o ilu_heap_relax_snode.o mark_relax.o mc64ad.o qselect.o lsame.o xerbla.o clacon.o scsum1.o icmax1.o slamch.o
ranlib /home/ahmad/SuperLU_4.3/lib/libsuperlu_4.3.a
ar cr /home/ahmad/SuperLU_4.3/lib/libsuperlu_4.3.a \
dcomplex.o zgssv.o zgssvx.o zsp_blas2.o zsp_blas3.o zgscon.o zlangs.o zgsequ.o zlaqgs.o zpivotgrowth.o zgsrfs.o zgstrf.o zgstrs.o zcopy_to_ucol.o zsnode_dfs.o zsnode_bmod.o zpanel_dfs.o zpanel_bmod.o zreadhb.o zreadrb.o zreadtriple.o zcolumn_dfs.o zcolumn_bmod.o zpivotL.o zpruneL.o zmemory.o zutil.o zmyblas2.o zgsisx.o zgsitrf.o zldperm.o ilu_zdrop_row.o ilu_zsnode_dfs.o ilu_zcolumn_dfs.o ilu_zpanel_dfs.o ilu_zcopy_to_ucol.o ilu_zpivotL.o zdiagonal.o superlu_timer.o util.o memory.o get_perm_c.o mmd.o sp_coletree.o sp_preorder.o sp_ienv.o relax_snode.o heap_relax_snode.o colamd.o ilu_relax_snode.o ilu_heap_relax_snode.o mark_relax.o mc64ad.o qselect.o lsame.o xerbla.o zlacon.o dzsum1.o izmax1.o dlamch.o
ranlib /home/ahmad/SuperLU_4.3/lib/libsuperlu_4.3.a
c

myzinsky, 2020-07-24:
Hello,

I don't see any problem here? 

Best,
Matthias

explorer1022, 2020-07-24:
Hi
I thought it uncompleted building.
Many thanks

closed 2020-07-25 by myzinsky