tukl-msd/DRAMSys #5: the server does not respose

explorer1022, 2020-07-26:
I tried the below commands but there was no response from the server. 

ahmad@ahmad-VirtualBox:~$ cd DRAMSys/DRAMSys/library/resources/configs/thermalsim
ahmad@ahmad-VirtualBox:~/DRAMSys/DRAMSys/library/resources/configs/thermalsim$  ~/3d-ice-latest/3d-ice-2.2.6/bin/3D-ICE-Server stack.stk 11880
Preparing stk data ... done !
Preparing thermal data ... done !
Creating socket ... done !
Waiting for client ...

lsteiner-tukl, 2020-07-27:
You have to start the 3D-ICE-Server and in another terminal run the DRAMSys simulation. Then you get the temperature and power map files as output (in the build/simulator directory).

myzinsky, 2020-07-27:
@explorer1022 please have a look on the readme: https://github.com/tukl-msd/DRAMSys#dramsys-with-thermal-simulation

explorer1022, 2020-07-27:
From where I can bring wideio-thermal.json because when I run below command it shows me bash: /DRAMSys: No such file or directory

ahmad@ahmad-VirtualBox:~$ cd DRAMSys/build/simulator/
ahmad@ahmad-VirtualBox:~/DRAMSys/build/simulator$ /DRAMSys ../../DRAMSys/library/resources/simulations/wideio-thermal.json
bash: /DRAMSys: No such file or directory

lsteiner-tukl, 2020-07-27:
@explorer1022 please check the readme carefully before you create an issue. A binary is executed with "./binary" and the readme also says "./DRAMSys ../../DRAMSys/library/resources/simulations/wideio-thermal.json" and not "/DRAMSys ../../DRAMSys/library/resources/simulations/wideio-thermal.json".

explorer1022, 2020-07-27:
when I run the below command I got 

 ahmad@ahmad-VirtualBox:~/DRAMSys/build/simulator$ ./DRAMSys ../../DRAMSys/library/resources/simulations/wideio-thermal.json

  |||
 +---+   Microelectronic Systems
=|   |=  Design Research Group
=|   |=  Technische Universität Kaiserslautern
 +---+  
  |||    DRAMSys4.0

Fatal: DRAMSys.TemperatureController: DRAMSys was build without support to dynamic temperature simulation. Check the README file for further details.
In file: /home/ahmad/DRAMSys/DRAMSys/library/src/simulation/../simulation/TemperatureController.h:78

Info: (I99) simulation aborted
Aborted (core dumped)

lsteiner-tukl, 2020-07-28:
The readme clearly says:

Export the environment variable LIBTHREED_ICE_HOME:
$ export LIBTHREED_ICE_HOME=${HOME}/3d-ice-latest/3d-ice-2.2.6

Running DRAMSys with Thermal Simulation
In order to run DRAMSys with thermal simulation you have to rerun CMake and rebuild the project.

closed 2020-07-28 by lsteiner-tukl

myzinsky, 2020-07-28:
@explorer1022 If you need help to setup DRAMSys on your site or for your project we can offer you a service to set it up according to your needs.