tukl-msd/DRAMSys #6: Usingj DRAMSys standalone

sherylll, 2020-07-29:
Hi I'd like to ask how DRAMSys should be included in my own project.

I added the following options to g++: `-L../../include -lDRAMSysLibrary` (include is where libDRAMSysLibrary.a resides)

It compiles fine. But adding `#include "DRAMSysLibrary.h"` in my scxx.cpp file results in not found error. Does the library happen to be called something else? Or is there other files I'm missing?

Many thanks!

myzinsky, 2020-07-29:
Are you sure that you have to correct paths? You also have to specify the path where the includes are with -I

lsteiner-tukl, 2020-07-29:
You have to include the file "DRAMSys/DRAMSys/library/src/simulation/DRAMSys.h". It is a bit confusing because the simulator binary is called DRAMSys and the library is called DRAMSysLibrary, however the header file with the class declaration is called DRAMSys.h.

sherylll, 2020-07-29:
Hi, thanks for your quick response, do you mean include path the DRAMSys repo (Ipath/to/DRAMSys) and putting `#include "DRAMSys/library/src/simulation/DRAMSys.h"` in cpp file?

lsteiner-tukl, 2020-07-29:
What do you mean by "include path the DRAMSys repo"? You have to include the DRAMSys.h where you want to use it and you have to make sure that the path to the file is correct.

sherylll, 2020-07-29:
Hi, maybe I miss understood...
But for a very simple case, I have only two things in a folder:
```
.
├── libDRAMSysLibrary.a
└── test.cpp
```
Within test.cpp:
```
#include "DRAMSys/DRAMSys/library/src/simulation/DRAMSys.h"
int main(){
    return 0;
}
```
And the output of compilation is:
```
g++ -o test test.cpp -L. -lDRAMSysLibrary
test.cpp:1:10: fatal error: DRAMSys/DRAMSys/library/src/simulation/DRAMSys.h: No such file or directory
 #include "DRAMSys/DRAMSys/library/src/simulation/DRAMSys.h"
          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
```

myzinsky, 2020-07-29:
Just 
‘‘‘C++
#include „DRAMSys.h“
‘‘‘

And the provide the absolute path to the header with ‘‘‘-I‘‘‘

lsteiner-tukl, 2020-07-29:
Actually it's not quite that easy. I have built DRAMSys as a library and created the test.cpp file inside the build folder where libDRAMSysLibrary.a is located. To compile the test:

g++ -std=c++11 test.cpp -o test -I../DRAMSys/library/src/simulation -L. -lDRAMSysLibrary -I$SYSTEMC_HOME/include -L$SYSTEMC_HOME/lib-linux64 -lsystemc -I../DRAMSys/library/src/common/third_party/DRAMPower/src -Lsrc/common/third_party/DRAMPower -lDRAMPower

In addition you need an sc_main inside your test file, mine was:

#include <systemc.h>
#include "DRAMSys.h"

int sc_main(int argc, char **argv)
{
    DRAMSys dramsys("test", "test", "test");
    return 0;
}

Please make sure that $SYSTEMC_HOME is defined or that you replace it with the right path.

sherylll, 2020-07-30:
Thanks, looks like it can work!
However there seem to be issue with SystemC version...
```
g++ -std=c++11 test.cpp -o test -I../DRAMSys/library/src/simulation -L. -lDRAMSysLibrary -I$SYSTEMC_HOME/include -L$SYSTEMC_HOME/lib-linux64 -lsystemc -I../DRAMSys/library/src/common/third_party/DRAMPower/src -Lsrc/common/third_party/DRAMPower -lDRAMPower

./libDRAMSysLibrary.a(DRAMSys.cpp.o): In function `_GLOBAL__sub_I__ZN7DRAMSysC2EN7sc_core14sc_module_nameENSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEES7_':
DRAMSys.cpp:(.text.startup+0x2c): undefined reference to `sc_core::sc_api_version_2_3_4_cxx201103L<&sc_core::SC_DISABLE_VIRTUAL_BIND_UNDEFINED_>::sc_api_version_2_3_4_cxx201103L(sc_core::sc_writer_policy, bool)'
./libDRAMSysLibrary.a(utils.cpp.o): In function `_GLOBAL__sub_I__ZN12TimeInterval16timeIsInIntervalEN7sc_core7sc_timeE':
utils.cpp:(.text.startup+0x2c): undefined reference to `sc_core::sc_api_version_2_3_4_cxx201103L<&sc_core::SC_DISABLE_VIRTUAL_BIND_UNDEFINED_>::sc_api_version_2_3_4_cxx201103L(sc_core::sc_writer_policy, bool)'
```
The latest systemc available on my system is 2.3.3, is there a way to tell make to use that version?

lsteiner-tukl, 2020-07-30:
It seems like your SystemC library has not been built with C++11 but with a different version. Try to leave out the -std=c++11 flag or use the -std=c++14 flag, this should be the default one for SystemC 2.3.4.

lsteiner-tukl, 2020-07-30:
Which SystemC library did you use to build the DRAMSys library? The submodule or your own local one? You can specify a local SystemC version with some environment variables (see readme). Make sure that you always link the same library. Also make sure to build both the library and your test program with the same C++ version (c++11 might not be the right one).

sherylll, 2020-07-30:
Hi, 
Thanks for pointing out, indeed I was using different systemc versions...
I set SYSTEMC_HOME variable and recompiled, still there are reference undefined errors occur
```
$ g++ -std=c++11 test.cpp -o test -I../DRAMSys/library/src/simulation -L. -lDRAMSysLibrary -I$SYSTEMC_HOME/include -L$SYSTEMC_HOME/lib-linux64 -lsystemc -I../DRAMSys/library/src/common/third_party/DRAMPower/src -Lsrc/common/third_party/DRAMPower -lDRAMPower
./libDRAMSysLibrary.a(DRAMSys.cpp.o): In function `__static_initialization_and_destruction_0(int, int)':
DRAMSys.cpp:(.text+0x2408): undefined reference to `sc_core::sc_api_version_2_3_3_cxx201103L<&sc_core::SC_DISABLE_VIRTUAL_BIND_UNDEFINED_>::sc_api_version_2_3_3_cxx201103L(sc_core::sc_writer_policy)'
./libDRAMSysLibrary.a(TemperatureController.cpp.o): In function `__static_initialization_and_destruction_0(int, int)':
TemperatureController.cpp:(.text+0xfd2): undefined reference to `sc_core::sc_api_version_2_3_3_cxx201103L<&sc_core::SC_DISABLE_VIRTUAL_BIND_UNDEFINED_>::sc_api_version_2_3_3_cxx201103L(sc_core::sc_writer_policy)'
./libDRAMSysLibrary.a(DramDDR3.cpp.o): In function `__static_initialization_and_destruction_0(int, int)':
DramDDR3.cpp:(.text+0xa31): undefined reference to `sc_core::sc_api_version_2_3_3_cxx201103L<&sc_core::SC_DISABLE_VIRTUAL_BIND_UNDEFINED_>::sc_api_version_2_3_3_cxx201103L(sc_core::sc_writer_policy)'
./libDRAMSysLibrary.a(DramDDR4.cpp.o): In function `__static_initialization_and_destruction_0(int, int)':
DramDDR4.cpp:(.text+0xa49): undefined reference to `sc_core::sc_api_version_2_3_3_cxx201103L<&sc_core::SC_DISABLE_VIRTUAL_BIND_UNDEFINED_>::sc_api_version_2_3_3_cxx201103L(sc_core::sc_writer_policy)'
./libDRAMSysLibrary.a(DramLPDDR4.cpp.o): In function `__static_initialization_and_destruction_0(int, int)':
DramLPDDR4.cpp:(.text+0x146): undefined reference to `sc_core::sc_api_version_2_3_3_cxx201103L<&sc_core::SC_DISABLE_VIRTUAL_BIND_UNDEFINED_>::sc_api_version_2_3_3_cxx201103L(sc_core::sc_writer_policy)'
./libDRAMSysLibrary.a(DramWideIO.cpp.o):DramWideIO.cpp:(.text+0x123a): more undefined references to `sc_core::sc_api_version_2_3_3_cxx201103L<&sc_core::SC_DISABLE_VIRTUAL_BIND_UNDEFINED_>::sc_api_version_2_3_3_cxx201103L(sc_core::sc_writer_policy)' follow
collect2: error: ld returned 1 exit status
``` 
To ensure that C++11 is used, I compiled with the CC and CXX variable set to `my/gcc -std=c++11` and `my/g++ -std=c++11` respectively.

lsteiner-tukl, 2020-07-30:
Are you sure that your local SystemC version was compiled with C++11? Because it seems that it was not. Can you check the output of 

nm -C <your SystemC library> | grep "sc_core::sc_api_version"

SystemC can be compiled with C++11 using the following configure options:

../configure CXXFLAGS="-DSC_CPLUSPLUS=201103L"

sherylll, 2020-07-30:
here is the output 
```
00000000000007b0 T sc_core::sc_api_version_2_3_3_cxx201402L<&sc_core::SC_DISABLE_VIRTUAL_BIND_UNDEFINED_>::sc_api_version_2_3_3_cxx201402L(sc_core::sc_writer_policy)
00000000000007b0 T sc_core::sc_api_version_2_3_3_cxx201402L<&sc_core::SC_DISABLE_VIRTUAL_BIND_UNDEFINED_>::sc_api_version_2_3_3_cxx201402L(sc_core::sc_writer_policy)
00000000000000a0 b sc_core::sc_api_version_2_3_3_cxx201402L<&sc_core::SC_DISABLE_VIRTUAL_BIND_UNDEFINED_>::sc_api_version_2_3_3_cxx201402L(sc_core::sc_writer_policy)::default_writer_policy_config
00000000000000a4 b sc_core::sc_api_version_2_3_3_cxx201402L<&sc_core::SC_DISABLE_VIRTUAL_BIND_UNDEFINED_>::sc_api_version_2_3_3_cxx201402L(sc_core::sc_writer_policy)::default_writer_policy_config_seen
```
yes I think that's where the problem is, my systemC was compiled using C++14

sherylll, 2020-07-30:
The test.cpp example works with the following, using systemc built by the project
```
g++ -std=c++11 test.cpp -o test -I../DRAMSys/library/src/simulation -L. -lDRAMSysLibrary -I../DRAMSys/library/src/common/third_party/systemc/src/ -Lsrc/common/third_party/systemc/src/ -lsystemc -I../DRAMSys/library/src/common/third_party/DRAMPower/src -Lsrc/common/third_party/DRAMPower -lDRAMPower
```
Maybe I'll recompile my systemc...
Thank you for your help!

closed 2020-07-30 by sherylll

lsteiner-tukl, 2020-07-30:
I think you can also build DRAMSys with C++14, that should be the easiest solution.