tukl-msd/DRAMSys #7: Execution fails - parsing problem

AntonioGenov, 2020-11-26:
Hello, I cloned the repository (with the -recursive command) and I have successfully built DRAMSys after adapting the CMAKE_CXX_STANDARD to my default one which is 14.

INFO: My compiler version is: gcc version 6.5.0 20181026 (Ubuntu 6.5.0-2ubuntu1~16.04)  -> so default C++ standard is C++14.

I tried to use the systemC version from library/third-party/.... and I also tried to use my two externally installed versions (2.3.2 and 2.3.3) by exporting the environment variables. Both ways I had the same problem during execution:

`~/DRAMSys/build/simulator$ ./DRAMSys ../../library/resources/simulations/ddr4-example.json 

■ ■ ■  DRAMSys4.0, Copyright (c) 2020
■ ■ ■  Technische Universitaet Kaiserslautern,
■ ■ ■  Fraunhofer IESE

Error while trying to parse file: ./../../DRAMSys/library/resources/configs/mcconfigs/fr_fcfs.json
message: [json.exception.parse_error.101] parse error at line 1, column 1: syntax error while parsing value - unexpected end of input; expected '[', '{', or a literal
terminate called after throwing an instance of 'nlohmann::detail::type_error'
  what():  [json.exception.type_error.305] cannot use operator[] with a string argument with number
Aborted (core dumped)`

No metter which .json file I take, the result is the same. 
From the message, I see that it may be a parsing problem, because it looks like it can not read the file (consider it like empty ? ).

Do you know what am I doing wrong ? 

Thanks!

lsteiner-tukl, 2020-11-26:
Please make sure that the build folder is located in the top directory of the repository (where the README.md is located). Then you should be able to run the simulation with

./DRAMSys ../../DRAMSys/library/resources/simulations/ddr4-example.json

AntonioGenov, 2020-11-27:
Okay, yes that was the problem. (a bit confusing with the two hierarchical DRAMSys directories.. my bad) 
Thanks !

Now it starts the simulation, but it terminates with an abort  (core dumped).

`
nxf44782@nxl34009-vm-11:~/Desktop/DRAMSys-clone/DRAMSys/build/simulator$ ./DRAMSys ../../DRAMSys/library/resources/simulations/ddr4-example.json   

■ ■ ■  DRAMSys4.0, Copyright (c) 2020
■ ■ ■  Technische Universitaet Kaiserslautern,
■ ■ ■  Fraunhofer IESE

===========================================================================
Used Address Mapping:

 Bg  1: 0000000000000000000000000000000000100000000000000000000000000000
 Bg  0: 0000000000000000000000000000000000010000000000000000000000000000
 Ba  1: 0000000000000000000000000000000010000000000000000000000000000000
 Ba  0: 0000000000000000000000000000000001000000000000000000000000000000
 Ro 14: 0000000000000000000000000000000000001000000000000000000000000000
 Ro 13: 0000000000000000000000000000000000000100000000000000000000000000
 Ro 12: 0000000000000000000000000000000000000010000000000000000000000000
 Ro 11: 0000000000000000000000000000000000000001000000000000000000000000
 Ro 10: 0000000000000000000000000000000000000000100000000000000000000000
 Ro  9: 0000000000000000000000000000000000000000010000000000000000000000
 Ro  8: 0000000000000000000000000000000000000000001000000000000000000000
 Ro  7: 0000000000000000000000000000000000000000000100000000000000000000
 Ro  6: 0000000000000000000000000000000000000000000010000000000000000000
 Ro  5: 0000000000000000000000000000000000000000000001000000000000000000
 Ro  4: 0000000000000000000000000000000000000000000000100000000000000000
 Ro  3: 0000000000000000000000000000000000000000000000010000000000000000
 Ro  2: 0000000000000000000000000000000000000000000000001000000000000000
 Ro  1: 0000000000000000000000000000000000000000000000000100000000000000
 Ro  0: 0000000000000000000000000000000000000000000000000010000000000000
 Co  9: 0000000000000000000000000000000000000000000000000001000000000000
 Co  8: 0000000000000000000000000000000000000000000000000000100000000000
 Co  7: 0000000000000000000000000000000000000000000000000000010000000000
 Co  6: 0000000000000000000000000000000000000000000000000000001000000000
 Co  5: 0000000000000000000000000000000000000000000000000000000100000000
 Co  4: 0000000000000000000000000000000000000000000000000000000010000000
 Co  3: 0000000000000000000000000000000000000000000000000000000001000000
 Co  2: 0000000000000000000000000000000000000000000000000000000000100000
 Co  1: 0000000000000000000000000000000000000000000000000000000000010000
 Co  0: 0000000000000000000000000000000000000000000000000000000000001000
 By  2: 0000000000000000000000000000000000000000000000000000000000000100
 By  1: 0000000000000000000000000000000000000000000000000000000000000010
 By  0: 0000000000000000000000000000000000000000000000000000000000000001

===========================================================================
Per Channel Configuration:

 Memory type:           DDR4
 Memory size in bytes:  4294967296
 Number of ranks:       1
 Number of bankgroups:  4
 Number of banks:       16
 Number of rows:        32768
 Number of columns:     1024
 Chip data bus width:   8
 Chip size in bits:     4294967296
 Chip Size in bytes:    536870912
 Devices/Chips on DIMM: 8

===========================================================================
terminate called after throwing an instance of 'sc_core::sc_report'
  what():  Error: /OSCI_TLM-2/multi_socket: DRAMSys.DRAMSys_tSocket: Call to nb_transport_fw without a registered callback for nb_transport_fw.
In file: ../../../src/tlm_utils/convenience_socket_bases.cpp:42
In process: ddr3_example_stl.fec_0 @ 0 s
Aborted (core dumped)
`

lsteiner-tukl, 2020-11-27:
Do you use the SystemC submodule or a different SystemC library?

AntonioGenov, 2020-11-27:
I was using the external one. 

I just tried with the submodule and it works with it.

However, I want to use it in association with another project and I would like to keep the external systemc ...
I tried it with 2.3.2 and 2.3.3 and had the same problem.

The makefile generation (cmake command), with the submodule took much more time.
Which is normal, considering that with my systemc version it was not able to find one directory, so it was just passing through there:

`
cc1plus: warning: /usr/local/systemc-2.3.2/include/tlm: not a directory
`
In the CMakeList of the DRAMSys/library it is indicated to search this directory when SYSTEMC_HOME is exported ...
and actually, from what I saw, this directory do not exist neither in my systemc version nor the submodule one. 

`
# Build:
target_include_directories(DRAMSysLibrary
    PUBLIC src/common/third_party/DRAMPower/src/
    PUBLIC $ENV{SYSTEMC_HOME}/include/
    PUBLIC $ENV{SYSTEMC_HOME}/include/tlm/
)
`

Do you know what can be the reason ? 
NOTE: Not sure if it is relevant, my systemc was built with the standard Makefile (not with CMake).

(Generally I don`t use CMake so sorry for these questions ...)

lsteiner-tukl, 2020-11-27:
The ```include/tlm``` directory only exists in the SystemC library of Synopsys, so you can ignore the warning. 

However, using a shared SystemC library is usually not a problem. Are you using the reference SystemC implementation from Accellera?

When you are building SystemC with CMake it also only creates a makefile. Did you run the configure script properly before running ```make``` (check the INSTALL.md of SystemC)? And did you set the right environment variables as described in the readme? 

Maybe download and build SystemC again and check if it works.

lsteiner-tukl, 2020-12-03:
@AntonioGenov Is it working now?

AntonioGenov, 2020-12-03:
Hello,
I downloaded the SystemC from the accellera website, so yes.  I made the default configuration, I just changed the C++ version to 14 instead of 98 and I changed the installation path prefix ... It should be good too ... 

I didn't had the time to try again.
However, I will now try to download it and rebuild it. 

I will close this "issue" because it was resolved and this one is no longer relevant to the current topic. 
If it doesn't work after the rebuild, I might open a new one. 

Thank you for your help!

closed 2020-12-03 by AntonioGenov