tukl-msd/DRAMSys #9: Gem5 and DRAMSys coupling error

AntonioGenov, 2021-01-22:
Hello, 

**Quick version:**

I am trying to simulate your builtin example with gem5, but I have the following error:
```
...
...
...
[ 93%] Built target DRAMSys
[ 95%] Building CXX object gem5/CMakeFiles/DRAMSys_gem5.dir/home/antonio/Softwares/gem5/gem5/util/tlm/src/sc_master_port.cc.o
cc1plus: warning: /usr/local/systemc2.4/include/tlm: not a directory
/home/antonio/Softwares/gem5/gem5/util/tlm/src/sc_master_port.cc: In member function 'Packet* Gem5SystemC::SCMasterPort::generatePacket(tlm::tlm_generic_payload&)':
/home/antonio/Softwares/gem5/gem5/util/tlm/src/sc_master_port.cc:50:15: error: 'class ExternalMaster' has no member named 'masterId'
   50 |         owner.masterId);
      |               ^~~~~~~~
make[2]: *** [gem5/CMakeFiles/DRAMSys_gem5.dir/build.make:134: gem5/CMakeFiles/DRAMSys_gem5.dir/home/antonio/Softwares/gem5/gem5/util/tlm/src/sc_master_port.cc.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:2524: gem5/CMakeFiles/DRAMSys_gem5.dir/all] Error 2
make: *** [Makefile:149: all] Error 2
```
Do you know what can be the reason? 

**Detailed version:**

Before arriving here, I had multiple problems.
First I downloaded the latest gem5 version (maybe not the commit that you adviced us in the README). I have followed the standard installation with gem5 from the google host.

When trying to create the gem5 library with the command (after the normal build of `build/ARM/gem5.opt`):
`scons --with-cxx-config --without-python --without-tcmalloc build/ARM/libgem5_opt.so` 

I had issue releated to a casting from Gem5ToTlmBridge to SimObject:
```
build/ARM/cxx_config/Gem5ToTlmBridge32.cc: In member function 'virtual SimObject* Gem5ToTlmBridge32CxxConfigParams::simObjectCreate()':
build/ARM/cxx_config/Gem5ToTlmBridge32.cc:136:25: error: cannot convert 'sc_gem5::Gem5ToTlmBridge<32>' to 'SimObject' in return
return this->create();
^
scons: *** [build/ARM/cxx_config/Gem5ToTlmBridge32.do] Error 1
```

After some research I found that with this command we use the gem5`s builtin SystemC and this is a problem.
 https://gem5.atlassian.net/browse/GEM5-639?jql=text%20~%20%22systemc%22 
We need to reuse gem5 as library in SystemC and not the contrary/default SystemC as a library in gem5 (logic). 
I had to add `USE_SYSTEMC=False` flag to this command in order to use my local systemC and it worked. (at least the library was generated).

Then, I had problems with the DRAMSys Cmake trying to fetch some source code from `gem5/utils/systemc/`, 
but in the current gem5 version there are not source file inside this directory. I needed to change it to `gem5/utils/systemc/gem5_within_systemc`.

Once done, I arrive to the current problem:

```
[ 93%] Built target DRAMSys
[ 95%] Building CXX object gem5/CMakeFiles/DRAMSys_gem5.dir/home/antonio/Softwares/gem5/gem5/util/tlm/src/sc_master_port.cc.o
cc1plus: warning: /usr/local/systemc2.4/include/tlm: not a directory
/home/antonio/Softwares/gem5/gem5/util/tlm/src/sc_master_port.cc: In member function 'Packet* Gem5SystemC::SCMasterPort::generatePacket(tlm::tlm_generic_payload&)':
/home/antonio/Softwares/gem5/gem5/util/tlm/src/sc_master_port.cc:50:15: error: 'class ExternalMaster' has no member named 'masterId'
   50 |         owner.masterId);
      |               ^~~~~~~~
make[2]: *** [gem5/CMakeFiles/DRAMSys_gem5.dir/build.make:134: gem5/CMakeFiles/DRAMSys_gem5.dir/home/antonio/Softwares/gem5/gem5/util/tlm/src/sc_master_port.cc.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:2524: gem5/CMakeFiles/DRAMSys_gem5.dir/all] Error 2
make: *** [Makefile:149: all] Error 2
```

lsteiner-tukl, 2021-01-22:
Hi,
the problem is that some changes in gem5 were introduced without adapting the SystemC and TLM coupling code. This is why we recommend to use commit a470ef5 in the readme.
However, since we are also currently trying to use the latest version of gem5 in combination with DRAMSys all those issues are known to me. The issues are already fixed on the develop branch of gem5, but there is now a new problem when building the library that is not yet fixed. So you should stick with the stable branch and add the fixes manually. Alternatively you can use our fork of the gem5 repo where the fixes are already applied:
https://github.com/tukl-msd/gem5
And as you already pointed out, you have to build the library with USE_SYSTEMC=False and adapt the paths in the DRAMSys CMake.

Here are all related Jira issues and merged fixes:
https://gem5.atlassian.net/browse/GEM5-874
https://gem5.atlassian.net/browse/GEM5-877
https://gem5.atlassian.net/browse/GEM5-880

https://gem5-review.googlesource.com/c/public/gem5/+/39215
https://gem5-review.googlesource.com/c/public/gem5/+/38997

lsteiner-tukl, 2021-02-10:
The develop branch of the official gem5 repo should also work now.

closed 2021-02-10 by lsteiner-tukl