tukl-msd/DRAMSys #10: Build Error

srivatsankrishnan, 2021-09-10:
Hey,
First of all, thanks for open sourcing the tool. I am trying to follow the instructions and running into a build error. I am not really sure what the issue is. I did use `--recursive` option during git clone.

```
[ 57%] Building CXX object library/CMakeFiles/DRAMSysLibrary.dir/src/common/DebugManager.cpp.o
In file included from /n/janapa_reddi_lab/skrishnan/workspace/DRAMSys/DRAMSys/library/src/common/DebugManager.cpp:37:0:
/n/janapa_reddi_lab/skrishnan/workspace/DRAMSys/DRAMSys/library/src/common/DebugManager.h:40:24: error: missing binary operator before token "("
 #if __has_cpp_attribute(maybe_unused)
```
Here is my OS configuration.

```
Distributor ID:	CentOS
Description:	CentOS Linux release 7.9.2009 (Core)
Release:	7.9.2009
Codename:	Core
```

My cmake version is `3.17.3`

lsteiner-tukl, 2021-09-13:
Hi,
can you try to replace the lines

```
#if __has_cpp_attribute(maybe_unused)
    #define NDEBUG_UNUSED(var_decl) [[maybe_unused]] var_decl
#elif (__GNUC__ || __clang__)
    #define NDEBUG_UNUSED(var_decl) var_decl __attribute__((unused))
#else
    #define	NDEBUG_UNUSED(var_decl) var_decl
#endif
```

with the line

`#define	NDEBUG_UNUSED(var_decl) var_decl`

in DRAMSys/DRAMSys/library/src/common/DebugManager.h?

srivatsankrishnan, 2021-09-13:
@lsteiner-tukl Thanks. This solves it.

closed 2021-09-13 by srivatsankrishnan