tukl-msd/DRAMSys #11: Command issued while in power-down mode.

srivatsankrishnan, 2021-09-16:
Hi,
When I try to run the DRAMSys, at times, I am get the following warning and the simulator terminates.
```
WARNING: Command issued while in power-down mode.
Command: 3, Timestamp: 75898, Bank: 0
```
Any idea how I should fix this? 

Thanks
Sri

lsteiner-tukl, 2021-09-17:
Hi,
this warning is not issued by DRAMSys but by the DRAMPower submodule. The problem is that DRAMPower currently does not support memory configurations with more than one rank per channel. However, the warning neither terminates the simulation nor influences the results of DRAMSys. Only the results of DRAMPower should not be trusted.
Regards

srivatsankrishnan, 2021-09-17:
Hi, Thanks for your response. I was also able to narrow it down to `PowerDownPolicy` setting I used. I was playing with this setting to see if it achieves lower power for some cases. If the field is set to `Staggered`, then at times it works and at times it throws that warning. It possible there is a bug (I will cross check) in my parsing the stdout, but it was returning a blank response. I was wondering if the rate at which the trace is fed could cause this warning.

```
"tracesetup": [
            {
                "clkMhz": 300,
                "name": "ddr3_example.stl"
            },
```

What would be the ideal setting to use the `Staggered` flag?

lsteiner-tukl, 2021-09-20:
Hi Sri,
as I mentioned the problem can occur when you simulate multi-rank configurations. Additionally, a warning is sometimes thrown when power down is entered by DRAMSys directly after a refresh command (although it is valid). As long as you do not need the power results everything is fine. However, if you want to simulate with DRAMPower make sure not to use multi-rank configurations. 
Regards

srivatsankrishnan, 2021-09-21:
Thanks for the information. I will keep that in mind. So basically I have to choose an adress mapping config that doesn't have an explicit rank  bits assigned? If my understanding is incorrect, can you please point me to a config file that doesn't use multi-rank configuration?

lsteiner-tukl, 2021-09-22:
The information about ranks is specified in the memspec files (located in DRAMSys/library/resources/configs/memspecs). The address mapping always has to fit the memspec (i.e., if you only have a single rank there should be no rank bit in the address mapping). But as I said, you only have to take care if you are interested in the results of DRAMPower.

closed 2021-09-28 by lsteiner-tukl