tukl-msd/DRAMSys #15: question about TraceAnalyzer and pre-load data to DRAMs

husterZC, 2022-10-04:
Hi,

DRAMsys4.0 is a very interesting project and I'm quite impressed it can simulate DRAM fastly. Apart from results on the terminal and a SQL database, I'm also pretty much interested in the TraceAnalyzer for visualization. Did you plan to open-source the TraceAnalyzer?

Besides, I noticed that each read/writes to DRAMs is through the TLM interface. Does DRAMsys4.0 support to pre-load data to DRAMs prior to simulation? Eg., pre-load an elf file into DRAMs.

lsteiner-tukl, 2022-10-05:
Hi,
the Trace Analyzer might be open sourced some time in the future. However, you can also get a license right now. Please contact [Matthias Jung](mailto:matthias.jung@iese.fraunhofer.de) for more information. 
Regarding your second question there is no such feature implemented, but you can easily implement it yourself. Data storage is managed in the file https://github.com/tukl-msd/DRAMSys/blob/master/DRAMSys/library/src/simulation/dram/Dram.cpp with a memory map or a simple dynamic array, which you could preload with some data before the simulation starts.

myzinsky, 2022-10-05:
@lsteiner-tukl should preload with ```transport_dbg()``` not also work?

lsteiner-tukl, 2022-10-05:
@myzinsky Sure, this would also work, but it is definitely more implementation effort.

closed 2022-11-10 by lsteiner-tukl