tukl-msd/DRAMSys #17: The Bandwidth calculation

Porcarar, 2022-10-09:
Hello, 
I have just started using DRAMSys and not familiar with all aspects yet. I have 2 questions and I hope you can help me with this. 
1. The bandwidth. I wanted to see how the bandwidth depends from clk freq. changes. I found only one clk parameter/variable located under "tracesetup"part. I tried different values for clkMhz in "tracesetup" but numbers for AVG BW, AVG BW\IDLE, MAX BW are always the same (I'm sure the config file path/name is correct).
Is there any other clk parameter that defines the transfer rate between controller and dram-device?
2. If there are 2 different configuration for tracesetup, how does this work?  
"tracesetup": [
            {
                "clkMhz": 300,
                "name": "ddr3_example.stl"
            },
            {
                "clkMhz": 400,
                "name": "ddr3_example.stl"
            }
Thank you.

myzinsky, 2022-10-09:
1. The DRAM clock rate is defined in the memspec files
2. Then you will have two trace players. One with 300 MHz and one with 400 MHz.

lsteiner-tukl, 2022-10-10:
As Matthias said the clock frequency between controller and DRAM is defined in the memspec files. But when you change the clock frequency you should also adapt the timing parameters accordingly because they are specified in clock cycles and most of them will increase with a higher frequency.

closed 2022-11-10 by lsteiner-tukl