tukl-msd/DRAMSys #19: Run error

cnjsdfcy, 2022-12-05:
Hi,

After compiled the project, the default config file "ddr5-example.json" was not found which caused a "**_Failed to open file_**" error.
After specified the config file to _DRAMSys/library/resources/simulations/ddr3-example.json_, another error came up: 
[image]

Not sure what's wrong with the json file's format, please help take a look, thanks.

lsteiner-tukl, 2022-12-05:
Hey,
as described in the readme you should call DRAMSys like this:
"
To run DRAMSys with a specific base config file:

$ ./DRAMSys ../../DRAMSys/library/resources/simulations/ddr3-example.json
To run DRAMSys with a specific base config file and a resource folder somewhere else to the standard:

$ ./DRAMSys ../../DRAMSys/tests/example_ddr3/simulations/ddr3-example.json ../../DRAMSys/tests/example_ddr3/
"
It seems that you did not navigate to the config file properly.

cnjsdfcy, 2022-12-05:
Ohh, I see, the second arg for resources path is the devil.
Thanks for you quickly help.
Closing.

closed 2022-12-05 by cnjsdfcy