tukl-msd/DRAMSys #20: one more increment of "nextChannelPayloadIDToAppend‘’ which should not be taken

gxflying, 2023-03-23:
Hi sir, I am learning about the Dramsys now,  
but when I read the source code of the controller, I found it appears to have a bug when dealing with the request from the frontend.

the bug occurs when a recieved request payload needs to be spilted into multiple children payload 

the "**nextChannelPayloadIDToAppend**" variable is increased two times for one request :  
   1. first increased in L642 in function **_createChildTranses_** which is called by **_manageRequests_** , 
   2. then increased again in L467 in manageRequests(), 
  
the final result is  that the  **nextChannelPayloadIDToAppend** **is increced by two for one request** !

I need your help to clarify it,  wish for your reply and thanks for your time~

https://github.com/tukl-msd/DRAMSys/blob/09f6dcbb91351e6ee7cadfc7bc8b29d97625db8f/DRAMSys/library/src/controller/Controller.cpp#L476
https://github.com/tukl-msd/DRAMSys/blob/09f6dcbb91351e6ee7cadfc7bc8b29d97625db8f/DRAMSys/library/src/controller/Controller.cpp#L642

lsteiner-tukl, 2023-03-23:
Hey, thanks for the hint. It seems that you are right, but we will have a look into it once more. However, the second increment should not influence the results of the simulation, as it is only important to guarantee a strict order within all requests.

lsteiner-tukl, 2023-03-24:
The bug has been fixed.

closed 2023-03-24 by lsteiner-tukl