tukl-msd/DRAMSys #23: clarify the point when manageResponses function is called

gxflying, 2023-04-22:
Hi sir,  when I learned about response flow of the controller in the source code of the Controller.cpp 
I found that the **manageResponses** will be called only when at the full cycle  (which is  the edge of the clock)
https://github.com/tukl-msd/DRAMSys/blob/7925e4a5e83bdd94d2b182a60ebcea616b50c69c/DRAMSys/library/src/controller/Controller.cpp#L257
so  the _transToRelease_ will be released at exactly the edge of a clock cycle,     
https://github.com/tukl-msd/DRAMSys/blob/7925e4a5e83bdd94d2b182a60ebcea616b50c69c/DRAMSys/library/src/controller/Controller.cpp#L496
but how to make sure  _transToRelease.time_ is at the edge of a clock cycle when it is calulated by :
      **transToRelease.time = sc_time_stamp() + delay;**   where sc_time_stamp() and delay can be any value?
https://github.com/tukl-msd/DRAMSys/blob/7925e4a5e83bdd94d2b182a60ebcea616b50c69c/DRAMSys/library/src/controller/Controller.cpp#L398

I  mean  how or where do we aligned the  **_transToRelease.time_**  to a full cycle (edge of a cycle) so that the transToRelease can be release ?   (from the code two conditions need to be meet to release the transToRelease : 1.  controllerMethod  is executed at the full cycle   2.  the transToRelease.time  is also a full cycle  , where full cycle means the edge of a clock)
https://github.com/tukl-msd/DRAMSys/blob/7925e4a5e83bdd94d2b182a60ebcea616b50c69c/DRAMSys/library/src/controller/Controller.cpp#L498

wish your reply 
thanks

derchr, 2023-04-24:
Hi,
in principle you are correct if we only analyze the Controller in isolation.
However, the Controller receives the END_RESP phases from the Arbiter. The Arbiter ensures that the `delay` lands exactly on a clock edge:
https://github.com/tukl-msd/DRAMSys/blob/7925e4a5e83bdd94d2b182a60ebcea616b50c69c/DRAMSys/library/src/simulation/Arbiter.cpp#L131-L132

gxflying, 2023-04-24:
@derchr thanks for your reply
the code snippet you referenced is in the nb_transport_fw interface of the arbiter, but I think the nb_transport_fw will be called from the master of the arbiter which may be a CPU to pass transactions to artiter and it is not use for the  END_RESP from arbiter to controller

the END_RESP phases from arbiter to Controller should be Arbiter.cpp#L186,  am I right?

and actually the value of delay passing from arbiter to ddr controller occurs when the nb_transport_fw of controller is called in ArbiterSimple::peqCallback,  and the delay will be added to _**transToRelease.time**_ directly in END_RESP flow in Controller.cpp#L398  (the nb_transport_fw should be implemented by a slave but be called by a master, so the controller's nv_transport_fw will be called by arbiter in Arbiter.cpp#L186 )
and this delay passed by arbiter can be any value !
and as in my experiment the simulation will never be finisded when the delay passed from arbiter is NOT at the edge of the tCK clk 
https://github.com/tukl-msd/DRAMSys/blob/7925e4a5e83bdd94d2b182a60ebcea616b50c69c/DRAMSys/library/src/controller/Controller.cpp#L398
https://github.com/tukl-msd/DRAMSys/blob/7925e4a5e83bdd94d2b182a60ebcea616b50c69c/DRAMSys/library/src/simulation/Arbiter.cpp#L186

https://github.com/tukl-msd/DRAMSys/blob/7925e4a5e83bdd94d2b182a60ebcea616b50c69c/DRAMSys/library/src/simulation/DRAMSys.cpp#L242

by the way,  
is it better that the actions take place inside the Controller to ensure the timing for controller? 
oherwise when other people want to integrate the Controller itself in their system with their own arbiter there may have problems as my experiment : simulation will never be finished

derchr, 2023-04-24:
> the END_RESP phases from arbiter to Controller should be Arbiter.cpp#L186, am I right?

No, it is in line https://github.com/tukl-msd/DRAMSys/blob/7925e4a5e83bdd94d2b182a60ebcea616b50c69c/DRAMSys/library/src/simulation/Arbiter.cpp#L235
There, no additional delay is added to the nb_transport_fw call.

When the master sends an END_RESP phase (which can be at any time), the Arbiter aligns the delay on whole clock cycles as I pointed out above. This delay is then passed to the PEQ of the Arbiter, which ensures that the peqCallback function of the Arbiter is called exactly at a clock edge.
Therefore, also the nb_transport_fw function of the Controller is called exactly at the clock edge with the passed delay of 0.

gxflying, 2023-04-24:
@derchr  thinks for your reply
I think I get the point in your reply  :  the BEGIN_REQ phase of a req sending to ddr controller is aligned to clock edge in arbiter, and the END_RESP phase to controller is always with SC_ZERO_TIME,    so all the outer trigger of the **_controllerMethod_** will be at tCK clock edge .  and the event trigger of controller itself is always at tCK clockedge too, so all the **_controllerMethod_** will be at tCK clock edge.  it makes sense.  and thank you for your clarification
one last question is : 
is it better that the actions are taken inside the Controller to ensure the timing for controller?
it helps when someone wants to integrate the controller in their system without the dramsys arbiter.

derchr, 2023-04-24:
Glad that I could help.

> is it better that the actions are taken inside the Controller to ensure the timing for controller?

While this is certainly possible, it's currently not one of our priorities. The Arbiter is, as the Controller, an integral part of DRAMSys and such a change would require architectural modifications.

closed 2023-04-25 by gxflying