tukl-msd/DRAMSys #25: Documentation for the DB API

junambi, 2023-06-26:
Hi,

Can you please point me to the list of API's that can be used to query the DB to better visualize the transactions going through the DRAMs.

Thank you,

lsteiner-tukl, 2023-06-26:
Hey,
as a starting point I recommend you to open the database with SQLiteBrowser and have a look at the structure. Then you can write your own SQL queries to calculate the desired metrics. A query for calculating the average response latency could look like this:
`SELECT AVG(Resp.PhaseBegin -  Req.PhaseBegin) / 1000 
        FROM Phases Req, Phases Resp 
        WHERE Req.PhaseName = 'REQ' AND Resp.PhaseName='RESP' AND Req.Transact = Resp.Transact
`
As you might know we also have our Trace Analyzer tool for visual analysis of the trace. If you interested in this tool please contact Matthias Jung.

junambi, 2023-06-27:
Hi Isteiner,

Thank you for the comments, i did reach out to Matthias, he mentioned that the Trace Analyzer is a private repo now, so, he suggested that i look at the DB directly and then get information from there.

That is the reason that i raised this query.

Thank you for the inputs.

I will try this and let you know.

Regards.

lsteiner-tukl, 2023-06-27:
Either you look at the DB directly or, as Matthias already mentioned, you can also have a license for the Trace Analyzer.

junambi, 2023-06-27:
Thank you for the comments, i will look directly at the DB for now and let you know if i have any questions.

Closing this for now.

closed 2023-06-27 by junambi