tukl-msd/DRAMSys #26: Fatal: Error in TraceRecorder: Error cannot open database

boobalan1996, 2023-07-10:
[image]
Please help me to fix this issue

lsteiner-tukl, 2023-07-10:
Hey,
can you check the permissions of the folder in which DRAMSys is executed? It seems that the write permissions are missing.

boobalan1996, 2023-07-11:
[image]

Write Permission is there, Could you please point out which database it's expecting?

lsteiner-tukl, 2023-07-11:
It is not expecting a database, it tries to create one. But the directory is owned by root. You need write permissions in the directory you are executing the binary from. E.g., if you execute the binary directly in build/bin/ you need write permissions in that folder.

boobalan1996, 2023-07-17:
Can we redirect the database creation from another directory?

lsteiner-tukl, 2023-07-17:
The database is created in the directory you are calling DRAMSys from. So you can choose whatever directory you want. Or you can disable the database creation completely in the simconfig.

boobalan1996, 2023-07-17:
[image]
Thanks ! 
I changed the root directory permission still facing different errors, Please check and help me to fix this.

lsteiner-tukl, 2023-07-17:
See #24

boobalan1996, 2023-07-17:
Thanks! I hope Highlighted warning is ignorable and the simulation is working fine. 
Could you please say how to analyze through simulation & trace analyzer Tools? 

[image]

lsteiner-tukl, 2023-07-17:
Yes, the warning is fine. A simulation creates an output SQLite trace database (.tdb) where all transactions and DRAM commands are recorded. You can either write some SQL queries to analyze the database or you can request a license for the Trace Analyzer, which provides a visualization of the database and some built-in metrics.

boobalan1996, 2023-08-08:
Hi,
I can see only address mapping & BW information in the terminal, 
Could you please say how I can check the log files about the working of memory and controller? 
To understand better do we have the option to enable log generation? If not can we write some function to generate logs?

lsteiner-tukl, 2023-08-08:
Could you please create a separate issue for this question? Thanks!

closed 2023-08-08 by lsteiner-tukl