tukl-msd/DRAMSys #29: Fatal: StlPlayer: Malformed trace file line 1.

KylePan-01, 2023-08-18:
Hi,
I'm new to DRAMSys and hope that I can get some guidances.
I created a trace file cifar_client.stl and run the following command: **./DRAMSys ../../configs/ddr4.json**. The pwd is DRAMSys/build/bin.
The _ddr4.json_ is shown as followed:
{
    "simulation": {
        "addressmapping": "am_ddr4_8x4Gbx8_dimm_p1KB_brc.json",
        "mcconfig": "fr_fcfs.json",
        "memspec": "JEDEC_4Gb_DDR4-1866_8bit_A.json",
        "simconfig": "sim1.json",
        "simulationid": "ddr4",
        "tracesetup": [
            {
                "clkMhz": 800,
                "name": "cifar_client.stl"
            }
        ]
    }
}

The _cifar_client.stl_ is shown as followed:
0:	Write	0x7ffde0689818
1:	Write	0x7ffde0689810
2:	Write	0x7ffde0689808
3:	Write	0x7ffde0689800
4:	Write	0x7ffde06897f8
5:	Write	0x7ffde06897f0
6:	Write	0x7ffde06897e8
7:	Write	0x7fec7438c5e0
8:	Read	0x7fec7438ce68
9:	Read	0x7fec7438d000
10:	Write	0x7fec7438d9f8
11:	Write	0x7fec7438d9e8
12:	Write	0x7fec7438da98
13:	Read	0x7fec7438ce78
14:	Write	0x7fec7438da48
15:	Read	0x7fec7438ce88
16:	Write	0x7fec7438dc88
17:	Read	0x7fec7438ce98
18:	Write	0x7fec7438da50
19:	Read	0x7fec7438cea8
20:	Write	0x7fec7438da58
...
31911729:	Read	0x7ffde0689278
31911730:	Read	0x7fec55fd0650
31911731:	Write	0x7fec55fd0938
31911732:	Read	0x7ffde0689270
31911733:	Read	0x7ffde0689248
31911734:	Read	0x7ffde0689288
31911735:	Read	0x7fec55fd0668
31911736:	Read	0x7ffde0689298
31911737:	Read	0x7ffde06892a0
31911738:	Read	0x7ffde06892a8
31911739:	Read	0x7ffde06892b0
31911740:	Read	0x7ffde06892b8
31911741:	Read	0x7ffde06892c0
31911742:	Read	0x7ffde06892c8
31911743:	Write	0x7ffde06892c8
31911744:	Read	0x7fec582dbe70

The output is:
■ ■ ■  DRAMSys5.0, Copyright (c) 2023
■ ■ ■  RPTU Kaiserslautern-Landau,
■ ■ ■  Fraunhofer IESE

===========================================================================
Memory Configuration:

 Memory type:           DDR4
 Memory size in bytes:  4294967296
 Channels:              1
 Ranks per channel:     1
 Bank groups per rank:  4
 Banks per rank:        16
 Rows per bank:         32768
 Columns per row:       1024
 Device width in bits:  8
 Device size in bits:   4294967296
 Device size in bytes:  536870912
 Devices per rank:      8

===========================================================================
Used Address Mapping:

 Bg  1: 0000000000000000000000000000000000100000000000000000000000000000
 Bg  0: 0000000000000000000000000000000000010000000000000000000000000000
 Ba  1: 0000000000000000000000000000000010000000000000000000000000000000
 Ba  0: 0000000000000000000000000000000001000000000000000000000000000000
 Ro 14: 0000000000000000000000000000000000001000000000000000000000000000
 Ro 13: 0000000000000000000000000000000000000100000000000000000000000000
 Ro 12: 0000000000000000000000000000000000000010000000000000000000000000
 Ro 11: 0000000000000000000000000000000000000001000000000000000000000000
 Ro 10: 0000000000000000000000000000000000000000100000000000000000000000
 Ro  9: 0000000000000000000000000000000000000000010000000000000000000000
 Ro  8: 0000000000000000000000000000000000000000001000000000000000000000
 Ro  7: 0000000000000000000000000000000000000000000100000000000000000000
 Ro  6: 0000000000000000000000000000000000000000000010000000000000000000
 Ro  5: 0000000000000000000000000000000000000000000001000000000000000000
 Ro  4: 0000000000000000000000000000000000000000000000100000000000000000
 Ro  3: 0000000000000000000000000000000000000000000000010000000000000000
 Ro  2: 0000000000000000000000000000000000000000000000001000000000000000
 Ro  1: 0000000000000000000000000000000000000000000000000100000000000000
 Ro  0: 0000000000000000000000000000000000000000000000000010000000000000
 Co  9: 0000000000000000000000000000000000000000000000000001000000000000
 Co  8: 0000000000000000000000000000000000000000000000000000100000000000
 Co  7: 0000000000000000000000000000000000000000000000000000010000000000
 Co  6: 0000000000000000000000000000000000000000000000000000001000000000
 Co  5: 0000000000000000000000000000000000000000000000000000000100000000
 Co  4: 0000000000000000000000000000000000000000000000000000000010000000
 Co  3: 0000000000000000000000000000000000000000000000000000000001000000
 Co  2: 0000000000000000000000000000000000000000000000000000000000100000
 Co  1: 0000000000000000000000000000000000000000000000000000000000010000
 Co  0: 0000000000000000000000000000000000000000000000000000000000001000
 By  2: 0000000000000000000000000000000000000000000000000000000000000100
 By  1: 0000000000000000000000000000000000000000000000000000000000000010
 By  0: 0000000000000000000000000000000000000000000000000000000000000001

===========================================================================

Fatal: StlPlayer: Malformed trace file line 1.
In file:DRAMSys/src/simulator/simulator/player/StlPlayer.cpp:180

Info: (I99) simulation aborted
Aborted

I have some difficulties on reading the StlPlayer.cpp to find out how to fix the failure. Could someone tell me where is the error and I want to know the **exact format of the trace file (xxx.stl)** in configs (P.S. the format in cifar_client.stl is _{Index:+\t+Write/Read+\t+Address(generated by Intel pintool)}_

Looking for anyone's reply!
Thanks

lsteiner-tukl, 2023-08-19:
The format of the trace file is described here: https://github.com/tukl-msd/DRAMSys/blob/master/configs/README.md
The first letter of read/write should be lowercase.

closed 2023-08-19 by myzinsky

KylePan-01, 2023-08-21:
Thanks a lot