tukl-msd/DRAMSys #31: Fatal: Configuration: Power analysis is only supported with included DRAMPower library!

KylePan-01, 2023-08-24:
Hi again!
I create my own trace file and modify the example.json in configs/simconfig, and then run the cmd:./DRAMSys ../../configs/ddr4-example.json
ddr4-example.json file:
{
    "simulation": {
        "addressmapping": "am_ddr4_8x4Gbx8_dimm_p1KB_brc.json",
        "mcconfig": "fr_fcfs.json",
        "memspec": "JEDEC_4Gb_DDR4-1866_8bit_A.json",
        "simconfig": "example.json",
        "simulationid": "ddr4-example",
        "tracesetup": [
            {
                "clkMhz": 800,
                "name": "ciafr.stl"
            }
        ]
    }
}

Then I got this failure infomation. I wonder how to use user-defined trace file in the DRAMSys?

Looking for your reply.
Thanks a lot!

lsteiner-tukl, 2023-08-24:
Which failure information?

lsteiner-tukl, 2023-08-24:
See #28

closed 2023-08-28 by KylePan-01