tukl-msd/DRAMSys #32: simulate single x8 DRAM

klaru, 2023-08-30:
How can I simulate a channel with a single x8 DRAM?

 I can simulate a channel with four or two DRAMs by changing the "nbrOfDevices" parameter in the memspec .json file AND simultaneously change the "BYTE_BIT" addresses by removing address 2 (for four x8 devices) or by removing addresses 1 and 2 (for two x8 devices).
However, this doesn't work for "nbrOfDevices"=1 since when I remove all "BYTE_BITS" I get a segmentation fault.

Also, how could I simulate a channel with a single x4 DRAM (which is less than a byte wide)?

lsteiner-tukl, 2023-08-30:
Thank you for the hint. Here is a patch that fixes this problem. However, simulating only one x4 device is not yet possible.
[fix_single_device.patch](https://github.com/tukl-msd/DRAMSys/files/12473664/fix_single_device.patch)

klaru, 2023-08-30:
Thanks for the patch. It works.

 

Any plans to make a single x4 device work as well?

 

Klaus

lsteiner-tukl, 2023-08-31:
One problem is that our address mappings work on byte granularity. Are you planning to run simulations with enabled storage? Because from a behavioral perspective it makes no difference whether you only simulate a single device or a whole DIMM.

closed 2026-04-20 by derchr