tukl-msd/DRAMSys #33: Build Error: dowmload fail

hk806569446, 2023-09-07:
It's a very interesting project for the DRAM simulation. Thank you for your support of the open source project. 
  Herein, I meet a error when I try to bulid DRAMSys.  it seems that the building process will connect "sqlite" website to download sqlite3. But my enviroment can't connect internet, is there any solution for it?

myzinsky, 2023-09-07:
You could for sure modify the cmake files and download the dependencies like sqlite and nlohman etc. manually. Please have a look in the cmake manuals.

closed 2023-09-07 by myzinsky

hk806569446, 2023-09-07:
> You could for sure modify the cmake files and download the dependencies like sqlite and nlohman etc. manually. Please have a look in the cmake manuals.

Thanks for your suggestion. I have solved it. But when I try to use "make"command. There is another problem below:
make[2]:*** [simulator/DRAMSys] Error 1
make[2]:*** [simulator/CMakeFiles/DRAMSys.dir/all] Error 2

The building process is 100%, But when I type command "./DRAMSys"  it shows "Command not found."

myzinsky, 2023-09-07:
You have to find the binary in the build folder, probably called simulation

hk806569446, 2023-09-07:
It seems that the binary "simulation" is not generated，I will try rebuild it using another compiler version. Thanks

lsteiner-tukl, 2023-09-07:
The binary is called DRAMSys and it should be located in build/src/.

hk806569446, 2023-09-07:
I am using DRAMSys4.0, because sqlite3 is intergrated in it.  In my directorty, there is not build/src/, I am not sure it is correct

lsteiner-tukl, 2023-09-07:
Ok, then you should follow the appropriate [readme](https://github.com/tukl-msd/DRAMSys/blob/v4.0/README.md).