tukl-msd/DRAMSys #35: ECC Controller mode in DRAMSys 5.0

Dang-Jie, 2023-10-14:
Hi, thanks for providing this open-source tool.

1. I have realized that there is an ECC Controller mode option in DRAMSys 4.0, but I am unable to find this option in the source code of DRAMSys 5.0. Could you please advise on how to enable ECC in version 5.0?

2. Also, there is an option called ErrorChipSeed in simconfig. Could you please explain how this option works? What value does the option in the example (e.g. 42) represent?
[image]

Thanks.

Dang-Jie, 2024-02-21:
Thanks for the reply. 
1. After i included the EccModule header file in the simulator.cpp, i should also include the field "ECCControllerMode: Enabled" in the simconfig right?

2. I understand, but I would like to know if the field itself is deprecated or just the value. Does this field affect the ECC mechanism?

3. And for the field "ErrorCSVFile", I assume that when it is blank, a CSV file containing Error data will be output to the default location. Am I right? If yes, where should i locate the file?

4. For #37 q1 (sorry for avoiding reopening the issue), where can I find the TLM debug transaction data? I tried enabling the 'Debug' option inside simconfig, but it didn't work. Can you provide me with details on how to enable TLM debug transactions during simulation?

lsteiner-tukl, 2024-02-22:
1. The ECCControllerMode is deprecated and does not have an effect any more. It is sufficient to add the EccModule in the simulator.cpp.
2. It does not affect anything anymore.
3. The file is not required for this "new" EccModule.
4. TLM debug transaction is not the same as a debug build. You should have a look at the SystemC TLM standard in order to understand what this means. The data is recorded in the variable "memory" inside the Dram class (src/libdramsys/DRAMSys/simulation/dram/Dram.h). You can also simply snoop this memory during the simulation.

Dang-Jie, 2024-03-28:
Hi 2 more questions
1. Can i get an example on how to call the ECC Module in simulator.cpp 
2. Can i get some insight on how the error (it is the payload?) is being generated in ECCModule? Is it a random generated or  generate by following some pattern?

Thanks