tukl-msd/DRAMSys #38: The process is terminated

KylePan-01, 2024-01-14:
Hi, I want to use DRAMSys to get the DRAM energy consumption of resnet18 training process, the command is **./DRAMSys ../../configs/ddr4-example.json**. The cfg file screenshot is 
[image]

And the output screeshot is 
[image]
[image]
I'm wondering if the problem is caused due to the size of the *resnet18.stl* is too large.
Looking for someone can help me,
Thanks alot!

lsteiner-tukl, 2024-01-15:
What is the size of the trace file? And can you find out at which position in the source code the program is failing by using a debugger?

KylePan-01, 2024-01-15:
Hi, the size of the trace file is 30GB and it is generated by Intel pintool, pinatrace.so; I'm sorry that I can't find the exception because I'm rookie for DRAMSys. 
I unsample the trace file for 100 times and get a 280MB trace file (read 100 traces and write 1 trace to a new file), then it works. And I'm wondering wether I can use the average power generated by the unsample trace file as the power of the original DRAM trace approximately?

lsteiner-tukl, 2024-01-16:
Can you give us some more insights on your chosen configuration? Is the option "DatabaseRecording" set to false or true in the simconfig?