tukl-msd/DRAMSys #39: DRAMPower does not support LPDDR4

Dang-Jie, 2024-03-05:
Hi, I am facing the following error when trying to simulate with LPDDR4

Fatal: DramLPDDR4: DRAMPower does not support LPDDR4
In file: /home/yijieleem/DRAMSys/src/libdramsys/DRAMSys/simulation/dram/DramLPDDR4.cpp:47

I used DRAMSys 5.0 and DRAMPower 5.0rc1
The configuration as below:
 "addressmapping": "am_lpddr4_8Gbx16_brc.json",
 "mcconfig": "fr_fcfs.json",
 "memspec": "JEDEC_8Gb_LPDDR4-3200_16bit.json",
 "simconfig": "example.json",
 
 everything remain defaulted except for the refresh policy inside fr_fcfs.json, which i changed to "PerBank"

lsteiner-tukl, 2024-03-18:
Hey, the problem is that DRAMSys still uses the older version of DRAMPower, which did not support LPDDR4. But you can try to integrate DRAMPower 5.0 into DRAMSys. We will also provide an update soon.

myzinsky, 2025-05-08:
Dear @jzhou1318 our team is just before the merge of the new DRAMPower with DRAMSys. This is just a matter of days (probably we can do it next week).

derchr, 2025-05-12:
Hi, LPDDR4 power simulation is now supported with the newest integration of DRAMPower. If you experience any issues with the new integration, feel free to open another issue.

closed 2025-05-12 by derchr