tukl-msd/DRAMSys #41: Could not run with the .stl trace files

ctobryan, 2024-05-23:
[image]
[image]

Hi, may I ask is there any additional requirement or dependencies set up in order to simulate DRAMSys using the provided trace files, example.stl?

myzinsky, 2024-05-23:
This might come from a malformed trace file (https://github.com/tukl-msd/DRAMSys/issues/29) or you are missing GIT LFS (https://github.com/tukl-msd/DRAMSys/issues/24)

ctobryan, 2024-05-23:
[image]
Thanks, the issue is solved.

myzinsky, 2024-05-23:
Looks good to me :-)

closed 2024-05-23 by myzinsky