tukl-msd/DRAMSys #44: Timing constraints for RFMab

hritwik567, 2024-08-03:
Hi, I am trying to understand the timing constraints for the RFM command. I was hoping to get some help.

Based on my understanding, this code does not check for timing constraints for RFMab, but it blocks **ALL** the banks if the RAA Counter of even **ONE** of the banks is greater than  RAAMMT.

Question: Is my understanding of the code correct?

https://github.com/tukl-msd/DRAMSys/blob/caf5845c79a4cd38635ffc1b6e316af65798e1d8/src/libdramsys/DRAMSys/controller/refresh/RefreshManagerAllBank.cpp#L142

lsteiner-tukl, 2024-08-05:
Yes, your understanding is correct. The standard says that if RAAMMT is reached, no additional ACT commands are allowed. In order to ensure that the refresh management command can be issued as early as possible, we block all bank machines. The timing constraints are checked in the respective timing checker, not the refresh manager.

hritwik567, 2024-08-05:
What happens if one bank's RAA counter is equal to RAAMMT and the timing constraints of RFMab are not satisfied yet? 

Will the Refresh manager block all the banks and wait for timing constraints to be satisfied?
Or will it allow other banks to receive ACTs and when the timing constraint is met, issue an RFMab?

I'm trying to understand what the standard means exactly. If RAAMMT is reached, are ACTs not allowed to that particular bank or all the banks?

I would assume ACTs are only blocked to a particular bank and the refresh manager can choose a policy for issuing RFMab. The 2 possible policies being
1. block all the banks and wait for timing constraints
2. block a particular bank, wait for timing constraints to meet and then launch RFMab

lsteiner-tukl, 2024-08-05:
It will block all banks and wait for timing constraints to be satisfied, yes. If you would allow ACT commands to be issued to other banks, the RFMab would be delayed because it can only be issued if all banks are precharged just like a REFab command. So your second policy does not make much sense.

hritwik567, 2024-08-05:
Understood.

I was just trying to figure out if the ACTs can be launched to other banks.

Thank you very much.