tukl-msd/DRAMSys #46: Compilation of TraceAnalyzer Fails under Windows OS

maestrosoft, 2024-08-21:
When TraceAnalyzer is built under Windows OS it fails with a linker error (see below). The problem is caused by missing the preprocessor definition QWT_DLL when the open-source of QT is used. I suspect a commercial license of QT will not have this problem since it is possible to link the application with a static QT library. An open-source QT doesn't install the static library since to link with a static library you need a commercial license. I cannot try it since I do not have a commercial license for QT. Thus some mean is needed to check if QT is commercial or open source and create the project file accordingly.

Automatic MOC and UIC for target TraceAnalyzer
mocs_compilation_Debug.obj : error LNK2001: unresolved external symbol "public: static struct QMetaObject const QwtPlot::staticMetaObject" (?staticMetaObject@QwtPlot@@2UQMetaObject@@B) [C:\git\D
RAMSys\BUILD\src\traceAnalyzer\TraceAnalyzer.vcxproj]
tracefiletab.obj : error LNK2001: unresolved external symbol "public: static struct QMetaObject const QwtPlotCanvas::staticMetaObject" (?staticMetaObject@QwtPlotCanvas@@2UQMetaObject@@B) [C:\git
\DRAMSys\BUILD\src\traceAnalyzer\TraceAnalyzer.vcxproj]
C:\git\DRAMSys\BUILD\bin\Debug\TraceAnalyzer.exe : fatal error LNK1120: 2 unresolved externals [C:\git\DRAMSys\BUILD\src\traceAnalyzer\TraceAnalyzer.vcxproj]

lsteiner-tukl, 2024-08-22:
We do not actually support Windows. But it seems to be an issue with Qwt. Did you install the Qwt library?

maestrosoft, 2024-08-22:
Yes I did. But when QT 5 is installed as an open source the static library of QT is not created. Only the DLL shared libraries. Thus when the Qwt is installed it detects that QT is open source and creates a DLL shared library. Not s static library. Under Window when you want to link against a shared library you need to explicitly export the symbols. For the Qwt library, this is done by defining QWT_DLL which must be added when TraceAnalyser is compiled.

I was able to successfully compile and run TraceAnalyzer once I added the preprocessor define QWT_DLL to the compilation.

Thus you'll probably need to modify CMakeList to detect if Qwt is compiled as a DLL or not which depends itself on the use of QT with a commercial license of open source without a license.