tukl-msd/DRAMSys #48: DRAMSys V5.0 integrate into GEM5 24.0.0.0 issue

idealgao, 2024-08-30:
Fully follow GEM5 DRAMSys (ext/dramsys/README) step, and sample test all passed.
build/X86/gem5.opt configs/example/gem5_library/dramsys/arm-hello-dramsys.py
build/X86/gem5.opt configs/example/gem5_library/dramsys/dramsys-traffic.py
build/X86/gem5.opt configs/example/dramsys.py

src/mem/DRAMSys.py
class DRAMSys(AbstractMemory):
type = "DRAMSys"
cxx_class = "gem5::memory::DRAMSys"
cxx_header = "mem/dramsys.hh"
tlm = TlmTargetSocket(32, "TLM target port")
#configuration = Param.String("Path to the DRAMSys configuration")
#resource_directory = Param.String("Path to the DRAMSys resource directory")
configuration = Param.String("**ext/dramsys/DRAMSys/configs/ddr4-example.json**")
resource_directory = Param.String("ext/dramsys/DRAMSys/configs")
recordable = Param.Bool(True, "Whether DRAMSys should record a trace file")

But when run se.py using follow command line, comes out errors
command line: build/X86/gem5.opt configs/deprecated/example/se.py --cpu-type=DerivO3CPU -n=2 --cpu-clock=3GHz --sys-clock=3GHz --caches --l1d_size=16kB --l1i_size=64kB --cacheline_size=64 --l1d_assoc=8 --l1i_assoc=8 --l2cache --num-l2caches=1 --l2_size=1MB --l2_assoc=16 --mem-channels=2 --mem-channels-intlv=64 --mem-ranks=1 --mem-type=DRAMSys --cmd=tests/test-progs/hello/bin/x86/linux/hello
warn: The se.py script is deprecated. It will be removed in future releases of  gem5.
AttributeError: object 'DRAMSys' has no attribute 'controller'
(C++ object is not yet constructed, so wrapped C++ methods are unavailable.)
At:
src/python/m5/SimObject.py(855): getattr
/home/idealgao/workshop/gem5/gem5_null/configs/common/MemConfig.py(243): config_mem
configs/deprecated/example/se.py(289): <module>
src/python/m5/main.py(669): main

BTW, if "--external-memory-system EXTERNAL_MEMORY_SYSTEM" added, the simulation will fail silently

derchr, 2024-09-16:
Hello,
the se.py script is not supported from our side as it is as it is deprecated. To use DRAMSys with gem5, you can use the new gem5 standard library.
The config `build/X86/gem5.opt configs/example/gem5_library/dramsys/arm-hello-dramsys.py` demonstrates how to use DRAMSys together with the gem5 standard library.
Let me know if you need further assistance.

idealgao, 2024-09-21:
Hi Derek, 

  Great thanks for your confirm. SE.py mode has been deprecated in GEM5, however I still wonder is there any possible to integrate the DRAMSys latest version into SE.py mode, using Gem5ToTlmBridge() or other wrapper? So, our project could shift to latest GEM5 + DRAMSys smoothly.

I have tried the README in V4.0 (https://github.com/tukl-msd/DRAMSys/blob/v4.0/DRAMSys/gem5/README.md), unfortunately, these steps do not work neither.

derchr, 2024-09-23:
> SE.py mode has been deprecated in GEM5, however I still wonder is there any possible to integrate the DRAMSys latest version into SE.py mode, using Gem5ToTlmBridge() or other wrapper? So, our project could shift to latest GEM5 + DRAMSys smoothly.

Yes, it should be possible. You can look at `configs/example/dramsys.py` and `src/python/gem5/components/memory/dramsys.py` on how it's done there with `Gem5ToTlmBridge`.

> I have tried the README in V4.0 (https://github.com/tukl-msd/DRAMSys/blob/v4.0/DRAMSys/gem5/README.md), unfortunately, these steps do not work neither.

The method described in https://github.com/tukl-msd/DRAMSys/blob/v4.0/DRAMSys/gem5/README.md is not maintained any more and will probably not work without adjustments.

Note that running gem5 in SE mode is fully supported with the standard library approach, it's just that the se.py script is deprecated.