tukl-msd/DRAMSys #50: Multiple tREFI for multiple channels of HBM2

harishprasad29, 2024-12-25:
Do we have a feature in DRAMSYS of having multiple tREFI for multiple channels in hbm2. In the memspec json files we have only one tREFI option which is used globally for all the channels. How to set different tREFI for different channels.

myzinsky, 2025-01-06:
This is certainly possible. However, this must be implemented in C++/SystemC-level very likely. Each channel has its own channel controller. I think the best way would be to make the changes there: 

https://github.com/tukl-msd/DRAMSys/tree/master/src/libdramsys/DRAMSys/controller/refresh

https://github.com/tukl-msd/DRAMSys/blob/master/src/libdramsys/DRAMSys/controller/Controller.cpp#L261

derchr, 2026-02-04:
The easiest way to achieve something like this currently, is to instantiate DRAMSys multiple times with different configurations. If you have any questions, do not hesitate to reopen.

closed 2026-02-04 by derchr