tukl-msd/DRAMSys #52: Multiple channels in hbm2

harishprasad29, 2025-01-23:
When I am simulating hbm2 multiple channels by adding channel bits in the address mapping json and changing number of channels in memspec json, the issuing of commands to different channels is not independent. Dramsys is not treating the channels independently and has a global total time variable for all the channels. Due to this the bandwidth utilization calculation is not proper. Can you look into this

derchr, 2025-01-23:
Hello, can you specify in more detail, what the issue is?

As both controllers/channels are part of the same simulation, both are simulated for the same simulation time. With two channels you get an output like this:

```
DRAMSys.controller0  Total Time:     12065 ns
DRAMSys.controller0  AVG BW:          21.39 Gb/s |   2.67 GB/s |   8.35 %
DRAMSys.controller0  AVG BW\IDLE:     29.83 Gb/s |   3.73 GB/s |  11.65 %
DRAMSys.controller0  MAX BW:         256.00 Gb/s |  32.00 GB/s | 100.00 %
DRAMSys.controller1  Total Time:     12065 ns
DRAMSys.controller1  AVG BW:          21.05 Gb/s |   2.63 GB/s |   8.22 %
DRAMSys.controller1  AVG BW\IDLE:     32.52 Gb/s |   4.06 GB/s |  12.70 %
DRAMSys.controller1  MAX BW:         256.00 Gb/s |  32.00 GB/s | 100.00 %
```

Here, the bandwidth is calculated by the occupation of the data bus of both independent channels and the total simulation time.

harishprasad29, 2025-01-24:
Thanks I got your point. What if I want to send the memory requests to all the channels at a time. Like if I have 2 independent channels, is it possible to give memory requests to them independently and concurrently at the start of the simulation from the arbiter module. At present the requests are sent sequentially to the channels right.

derchr, 2025-01-24:
No, the channels are already accessed concurrently. The address together with the address mapping defines to which channel a memory request is sent to. If you have two requests issued consecutively to two different channels, then both channel controllers independently handle the two accesses.